ROBOTIS-GIT/ROBOTIS-OP3 #9: Migrating to catkin build

yoshimalucky, 2017-09-25:
I sent some PR to fix the build errors of the OP3 packages with catkin build. catkin build is a very common tool which is widely used in the ROS community, so I think we need to make some migration process for our OP3 packages.

Read "Migration Troubleshooting". 
http://catkin-tools.readthedocs.io/en/latest/migration.html

ROBOTIS-Kayman, 2017-10-12:
[image] [유지보수 - Github Issue / Migrating from catkin_make to catkin build](https://trello.com/c/4UW50H90/7-%EC%9C%A0%EC%A7%80%EB%B3%B4%EC%88%98-github-issue-migrating-from-catkinmake-to-catkin-build)

robotpilot, 2018-04-24:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thank you for your consideration and supports.

closed 2018-04-24 by robotpilot