ROBOTIS-GIT/ROBOTIS-OP3 #12: How to initialize the dynamixel motor settings

yoshimalucky, 2017-10-13:
Accidentally, I overwrote the servo motor settings with this script:
https://github.com/ROBOTIS-JAPAN-GIT/dynamixel_mikata_arm/blob/master/install.sh
https://github.com/ROBOTIS-JAPAN-GIT/dynamixel_mikata_arm/blob/master/mikata_arm_toolbox/src/dxl_setup.cpp

The main cause seems to be the second program. This program uses dxl_writeAll and dxl_write to change the servo motor settings, so it should overwrite the default motor settings.

Does any of the OP3 repositories have this kind of script which initialize the servo motor setting? Or, is there any document which explain about the parameters which should be set for the motors?

ROBOTIS-Kayman, 2017-10-16:
In the case of OP3, when ```op3_manager``` starts, dynamixel settings are loaded from yaml file.
(dxl_init yaml : https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/dxl_init_OP3.yaml)

the format of yaml is like below.
```
joint_name_1 : 
  control_table_item : value
  control_table_item : value
  ...
joint_name_2 : 
  control_table_item : value
  control_table_item : value
  ...
```

you can refer to [the device file](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/master/robotis_device/devices/dynamixel/XM-430-W350.device) about the control_table_item.

yoshimalucky, 2017-10-17:
I could fix OP3. Thanks.

closed 2017-10-17 by yoshimalucky