ROBOTIS-GIT/ROBOTIS-OP3 #39: Undefined reference to octomap when catkin_make

arixrobotics, 2018-05-29:
I'm having undefined reference error again when running catkin_make. This one looks like to octomap library:

> [ 94%] Built target open_cr_module
[ 95%] Built target op3_online_walking_module
[ 97%] Linking CXX executable /home/arif/catkin_ws/devel/lib/humanoid_localization/localization_node
[ 97%] Built target op3_action_editor
[ 98%] Built target offset_tuner_server_node
[100%] Built target op3_manager
/home/arif/catkin_ws/devel/lib/libmapmodel.so: undefined reference to `octomap::AbstractOccupancyOcTree::writeBinary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/home/arif/catkin_ws/devel/lib/libmapmodel.so: undefined reference to `octomap::OcTree::OcTree(double)'
/home/arif/catkin_ws/devel/lib/libmapmodel.so: undefined reference to `octomap::ColorOcTree::ColorOcTree(double)'
/home/arif/catkin_ws/devel/lib/libmapmodel.so: undefined reference to `octomath::operator<<(std::ostream&, octomath::Vector3 const&)'
/home/arif/catkin_ws/devel/lib/libmapmodel.so: undefined reference to `octomap::AbstractOcTree::read(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
humanoid_navigation/humanoid_localization/CMakeFiles/localization_node.dir/build.make:319: recipe for target '/home/arif/catkin_ws/devel/lib/humanoid_localization/localization_node' failed
make[2]: *** [/home/arif/catkin_ws/devel/lib/humanoid_localization/localization_node] Error 1
CMakeFiles/Makefile2:13252: recipe for target 'humanoid_navigation/humanoid_localization/CMakeFiles/localization_node.dir/all' failed
make[1]: *** [humanoid_navigation/humanoid_localization/CMakeFiles/localization_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Any idea on how to solve this?

ROBOTIS-Kayman, 2018-05-29:
I think you didn't install prerequisite packages for ROBOTIS-OP3 ros packages.
refer to the [link(emanual)](http://emanual.robotis.com/docs/en/platform/op3/recovery/#installing-ros-packages-for-robotis-op3).

arixrobotics, 2018-05-29:
I have followed the installation procedure closely. 

Anyway, as usual, after posting the issue, the error magically goes away.... 

Thanks for your time.

closed 2018-05-29 by arixrobotics