ROBOTIS-GIT/ROBOTIS-OP3 #40: Disable upper body control when walking

arixrobotics, 2018-05-30:
Hi, 
I'm trying to do the following:
Step 1: OP3 raises its hands using Action Module
Step 2: OP3 walks forward using Walking Module

Raising the hand in Step 1 is ok. 
But, when I enable walking module (by publishing "walking_module" to /robotis/enable_ctrl_module), it will cause the hands to go down so that the robot is in walk init position. 

So the question is, how can I enable walking module, without affecting the robot's upper body (hands & head)? I vaguely remember that for Darwin and OP2, we can do this by setting the servo offset to a very high number? How can I do this for OP3?

Thanks, 
Arif

ROBOTIS-Kayman, 2018-05-30:
In order to play the motion and make OP3 walk, each joint has to be set with specific modules that control joints.
There are two methods to set modules on joints;

- The first method configures predefined set of joints with a specific module. You can use either the topic or the service for this method as shown below.
  - topic name : /robotis/enable_ctrl_module
  - service name : /set_present_ctrl_modules
- The second method sets each joint with specific modules separately. You can also use the topic or the service for this method.
  - topic name : /robotis/set_joint_ctrl_modules
  - service name : /robotis/set_present_joint_ctrl_modules

You might want to use the second method to control each joint as you wish.
Please refer to the below link to see the example of how to set different modules on the head and the body.
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Demo/blob/952d9c42055c52e3c29b9db22cec8dc5631982ad/op3_demo/src/soccer/soccer_demo.cpp#L222

robotpilot, 2018-06-06:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

arixrobotics, 2018-06-10:
Hi, 
Sorry for the delay. I've managed to test the method suggested above, and I can confirm that it works. 

However, there seems to be a weird behavior for the ankles. Whenever I use /robotis/set_joint_ctrl_modules to set them into walking module, one of them moves into a weird position, somehow. 

This video should explain the problem better:
https://youtu.be/EStDLiBS500

This is the command that I ran:
`$ rostopic pub /robotis/set_joint_ctrl_modules robotis_controller_msgs/JointCtrldule '{joint_name: [l_ank_pitch, r_ank_pitch], module_name: [walking_module, walking_module]}'`

Hope you could check whether this is a bug in the code, or its due to some of my settings. And hopefully solve the problem too. 

Arif

arixrobotics, 2018-06-11:
The same ankle behaviour is observed when setting the module via service /robotis/set_present_joint_ctrl_modules.

ROBOTIS-Kayman, 2018-06-12:
Hi, @arixrobotics
Thank you for your contribution.
I'll check about this issue and answer to you.

arixrobotics, 2018-06-25:
Hi @ROBOTIS-Kayman, 
Any update on this? Did you manage to replicate?

ROBOTIS-Kayman, 2018-06-26:
@arixrobotics 
I'm sorry I'm late. I was on business trip for 2 weeks.
I did not start this problem. Another job is scheduled for now.
I think I can start this issue after that.

When I reproduce this condition, I will let you know.

arixrobotics, 2018-06-27:
As soon as you can please. Our client is waiting too :)

robotpilot, 2018-07-15:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue.

ROBOTIS-Kayman, 2018-07-16:
Hi @arixrobotics 
I'm sorry I'm late. We found the cause of this error. It's a initialize problem.
Pull the master branch and merge with it. then this issue will be gone.

arixrobotics, 2018-07-16:
Cool! 
Unfortunately I don't have access to the robot anymore. I'll ask my client to test it out, in the mean time, let's consider this closed. 
Thank you for your help!

closed 2018-07-16 by arixrobotics