ROBOTIS-GIT/ROBOTIS-OP3 #41: Error First Bulk Read

haritsahm, 2018-05-31:
I can't start the manager every time i add the opencr as sensor because i got this first bulk read fail.
but if i disable the opencr in op.robot, everything works

[image]

ROBOTIS-Kayman, 2018-06-01:
I can't figure it out what went wrong.

Would you give me some extra information like .robot file?
What protocol version of MX-28 and MX-64 do you use?

robotpilot, 2018-06-07:
his issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

haritsahm, 2018-06-09:
sorry i was busy, i would like to reopen this issue, this is my setup and opencr debug 
[image]

[image]

ROBOTIS-zerom, 2018-06-12:
In ROBOTIS Framework, only one protocol can be used for one port.

Change the MX-28 and MX-64 firmware to 2.0. 
1. Download R+ Manger 2.0 from here : http://en.robotis.com/service/downloadpage.php?ca_id=10 
2. Recovery the MX-28 and MX-64 firmware. : http://emanual.robotis.com/docs/en/software/rplus2/manager/#firmware-recovery

then all devices will use protocol 2.0

MX-28 (2.0) manual : http://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/
MX-64 (2.0) manual : http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/

haritsahm, 2018-06-12:
oh okay i will try it later! i thought it could use multiple protocol

robotpilot, 2018-06-20:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2018-06-20 by robotpilot

wiseenu, 2022-10-27:
i have same problem with error first bulk read. this my pict of the device
[image]