ROBOTIS-GIT/ROBOTIS-OP3 #43: op3 simulation not responding

hayday100, 2018-06-21:
I am trying the op3 Gazebo simulation.  I started roscore in one terminal.  
In the rest three terminals, I need to use Python2.7 virtualenv and run the roslaunch commands and attached the output.  Other than some warnings (... for same type) and serial port error (which should not affect a simulation), I don't see anything seriously wrong.  They all talk to the same master node on local port 11311.  But the issue is the simulation is not responding to any demo buttons.  I had the simulation worked once but now it no longer works and I don't know how to get it to work again.  Please help!  Thanks! 

#in second terminal
roslaunch op3_gazebo robotis_world.launch
`xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://home-desktop:46235/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robotis_op3/head_pan_position/joint: head_pan
 * /robotis_op3/head_pan_position/pid/d: 0.0
 * /robotis_op3/head_pan_position/pid/i: 0.0
 * /robotis_op3/head_pan_position/pid/p: 5.0
 * /robotis_op3/head_pan_position/type: effort_controller...
 * /robotis_op3/head_tilt_position/joint: head_tilt
 * /robotis_op3/head_tilt_position/pid/d: 0.0
 * /robotis_op3/head_tilt_position/pid/i: 0.0
 * /robotis_op3/head_tilt_position/pid/p: 5.0
 * /robotis_op3/head_tilt_position/type: effort_controller...
 * /robotis_op3/joint_state_controller/publish_rate: 50
 * /robotis_op3/joint_state_controller/type: joint_state_contr...
 * /robotis_op3/l_ank_pitch_position/joint: l_ank_pitch
 * /robotis_op3/l_ank_pitch_position/pid/d: 0.0
 * /robotis_op3/l_ank_pitch_position/pid/i: 0.0
 * /robotis_op3/l_ank_pitch_position/pid/p: 50.0
 * /robotis_op3/l_ank_pitch_position/type: effort_controller...
 * /robotis_op3/l_ank_roll_position/joint: l_ank_roll
 * /robotis_op3/l_ank_roll_position/pid/d: 0.0
 * /robotis_op3/l_ank_roll_position/pid/i: 0.0
 * /robotis_op3/l_ank_roll_position/pid/p: 50.0
 * /robotis_op3/l_ank_roll_position/type: effort_controller...
 * /robotis_op3/l_el_position/joint: l_el
 * /robotis_op3/l_el_position/pid/d: 0.0
 * /robotis_op3/l_el_position/pid/i: 0.0
 * /robotis_op3/l_el_position/pid/p: 10.0
 * /robotis_op3/l_el_position/type: effort_controller...
 * /robotis_op3/l_hip_pitch_position/joint: l_hip_pitch
 * /robotis_op3/l_hip_pitch_position/pid/d: 0.0
 * /robotis_op3/l_hip_pitch_position/pid/i: 0.0
 * /robotis_op3/l_hip_pitch_position/pid/p: 50.0
 * /robotis_op3/l_hip_pitch_position/type: effort_controller...
 * /robotis_op3/l_hip_roll_position/joint: l_hip_roll
 * /robotis_op3/l_hip_roll_position/pid/d: 0.0
 * /robotis_op3/l_hip_roll_position/pid/i: 0.0
 * /robotis_op3/l_hip_roll_position/pid/p: 50.0
 * /robotis_op3/l_hip_roll_position/type: effort_controller...
 * /robotis_op3/l_hip_yaw_position/joint: l_hip_yaw
 * /robotis_op3/l_hip_yaw_position/pid/d: 0.0
 * /robotis_op3/l_hip_yaw_position/pid/i: 0.0
 * /robotis_op3/l_hip_yaw_position/pid/p: 50.0
 * /robotis_op3/l_hip_yaw_position/type: effort_controller...
 * /robotis_op3/l_knee_position/joint: l_knee
 * /robotis_op3/l_knee_position/pid/d: 0.0
 * /robotis_op3/l_knee_position/pid/i: 0.0
 * /robotis_op3/l_knee_position/pid/p: 50.0
 * /robotis_op3/l_knee_position/type: effort_controller...
 * /robotis_op3/l_sho_pitch_position/joint: l_sho_pitch
 * /robotis_op3/l_sho_pitch_position/pid/d: 0.0
 * /robotis_op3/l_sho_pitch_position/pid/i: 0.0
 * /robotis_op3/l_sho_pitch_position/pid/p: 10.0
 * /robotis_op3/l_sho_pitch_position/type: effort_controller...
 * /robotis_op3/l_sho_roll_position/joint: l_sho_roll
 * /robotis_op3/l_sho_roll_position/pid/d: 0.0
[... 159 lines cut ...]
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://home-desktop:42078/

SUMMARY
========

PARAMETERS
 * /demo_config: /home/chen/catkin...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    op3_demo_opc (op3_gui_demo/op3_gui_demo)

ROS_MASTER_URI=http://home-desktop:11311/

process[op3_demo_opc-1]: started with pid [7348]
[ INFO] [1529578380.011487246]: Initialized node handle for default demo
[ INFO] [1529578380.023344695]: Initialized node handle for preview walking
[ INFO] [1529578414.106802597]: Go to robot initial pose.
[ INFO] [1529578420.400453653]: Get current Mode
[ INFO] [1529578427.454093042]: Set Mode : head_control_module
`

arixrobotics, 2018-06-25:
Just to be sure, have you click the Start button in Gazebo?

ROBOTIS-Kayman, 2018-06-27:
I'm sorry. but I can't figure out what the problem is.
Could you provide me more information such as screenshots and the button you clicked on?

arixrobotics, 2018-06-27:
In the screenshot you @hayday100  provided, the start button on Gazebo is not pressed. Try clicking on it. Then only the simulation will start.

hayday100, 2018-06-27:
[image]
Thank you and I cannot believe that I ignored the Play button so many times.  Now that I pushed the Play button, the OP3 is responding but it will immediately lay down on its back and then stopped responding to the buttons again.  I tried the "get up (front)" and get up (back)" among other buttons but the simulation is stuck over there.  All programs are still running.  The side pane messages in the demo has "Succeed to start page 123", "failed to start page 122", etc.

ROBOTIS-Kayman, 2018-06-28:
I think there are no controllers for gazebo.
Did you install following packages?
```
# apt-get install ros-kinetic-ros-control
# apt-get install ros-kinetic-ros-controllers
# apt-get install ros-kinetic-gazebo-ros-control
```

hayday100, 2018-06-28:
Thank you.  That's what the problem is.  In the process of installing ros-kinetic-gazebo-ros-control I had to downgrade several libgazebo9 packages to libgazebo7.  Not that the version difference matters to me, I just wonder if it is because libgazebo9 has some issues with ros-kinetic-gazebo-ros-control which caused the broken packages issues I had.  I did some experiments with realtime camera video transfer after the last experiment with OP3 which might have caused the package issues.  Anyway, once problem at a time.  
The next question I have is a easier way to restart the robot simulation from initial pose after it falls from walking.  When OP3 falls face down, the "get up front" or "get up back" won't get it up.  So I have to restart gazebo, op3_manager and op3_demo all over right now.  And I also happen to run into another position when the robot becomes a box and get stuck.
[image]

ROBOTIS-Kayman, 2018-07-02:
- easier way to restart the robot simulation from initial pose after it falls from walking
How about Using `Reset Model Poses`(Edit -> Reset Model Poses) instead of restarting gazebo.

 - becoming a "box"
I think when OP3 is getting up, some friction problem is happened. I also do not know much about this issue. I'm sorry I can not help you any more.

hayday100, 2018-07-03:
It sounds like friction is a very common problem in simulators which give the robots difficulty to walk stably in a virtual environment.  The community NAO gazebo simulator seems to share the same problem.  When it comes to the comparison of OP3 with NAO, your strength is being open-source.  However, your robot is more expensive and there is no Python API for the time being.  OP3 gripping hands will add extra cost while NAO has gripping hands coming with it.   Any insights?

robotpilot, 2018-07-09:
Hi @hayday100,

Thank you for your consideration and supports. I would greatly appreciate it. Your comments are a great help to us, and your interest and support will allow us to grow an open project that we are working together. We will take the feedback of you and other customers and will do our best to make better products and support you. 

I'm going to close this until you update the issue with more information. You can reopen this issue to show this issue to the users whenever. Once again, thank you for your help and supports. :)

closed 2018-07-09 by robotpilot