ROBOTIS-GIT/ROBOTIS-OP3 #44: robotis_op.urdf.xacro error

hayday100, 2018-06-24:
This error is over the OP2 simulation.  The catkin_make was error-free.  But when trying to launch it with

`roslaunch robotis_op_gazebo robotis_op_gazebo_position_control_empty_world.launch`

It give the following error.  Some similar discussion online suggest there might be a non-ASCII character in the xacro file but I don't see any in gedit and manually download the file again doesn't make a difference so I don't know how to fix it.  

```
Property names must be valid python identifiers: lb-in-in_kg-m-m None None
when processing file: /home/chen/catkin_ws/src/robotis_op_common/robotis_op_description/robots/robotis_op.urdf.xacro
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/chen/catkin_ws/src/robotis_op_common/robotis_op_description/robots/robotis_op.urdf.xacro'] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find robotis_op_description)/robots/robotis_op.urdf.xacro'" name="robot_description"/>
The traceback for the exception was written to the log file
```
I was trying to locate the roslaunch log file without luck.  If you need the log file, just let me know where it is and I will upload it.
```
cd $ROS_ROOT/log
bash: cd: /opt/ros/kinetic/share/ros/log: No such file or directory
```

ROBOTIS-Kayman, 2018-06-27:
Where did you get the robotis op2 simulator?
We do not officially provide the robotis op2 simulator.
So I have nothing to be able to advice this issue.

hayday100, 2018-06-27:
I followed instructions at:
https://github.com/ROBOTIS-OP/robotis_op_documentation/wiki/Installation-on--PC

Sorry if OP2 has nothing to do with your business.

ROBOTIS-Kayman, 2018-06-28:
OP2 is also related to my business.
However, this package is not officially provided by ROBOTIS. I do not know about the package.
So it is difficult to answer your question.

It's scheduled to release official ROBOTIS-OP2 ROS packages within a month.
Please wait for release.

hayday100, 2018-06-28:
Thank you.  Let me know when it is released.  I wish to compare the two models in simulation.

closed 2018-07-02 by robotpilot