ROBOTIS-GIT/ROBOTIS-OP3 #45: Catkin_make Failed

idrobotiskidslab, 2018-06-30:
please help me. Why should I do ? Thank You
[image]

ROBOTIS-Kayman, 2018-07-02:
Did you install `octomap` package and related packages?
```
$ sudo apt-get install ros-kinetic-map-server
$ sudo apt-get install ros-kinetic-humanoid-nav-msgs
$ sudo apt-get install ros-kinetic-nav-msgs
$ sudo apt-get install ros-kinetic-octomap
$ sudo apt-get install ros-kinetic-octomap-msgs
$ sudo apt-get install ros-kinetic-octomap-ros
$ sudo apt-get install ros-kinetic-octomap-server
```

Please refer to [e-manual](http://emanual.robotis.com/docs/en/platform/op3/recovery/#installing-ros-packages-for-robotis-op3).

idrobotiskidslab, 2018-07-02:
thank you for your reply.
I have install that related packages.
after I type `git checkout -b kinetic` in sbpl directory and then catkin_make has been success.
But when I launch Bring_up, I get Error again like this
[image]

ROBOTIS-Kayman, 2018-07-03:
I think there is a problem with the right arm.
Check the actuators at right shoulder pitch(ID 1) and right shoulder roll(ID 3) using [dynamixel manager](http://emanual.robotis.com/docs/en/software/rplus2/manager/) or [dynamixel workbench](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#introduction).

If some problem has found, follow the AS process.

idrobotiskidslab, 2018-07-03:
How about with the error **camera.yaml not found** ?

ROBOTIS-Kayman, 2018-07-03:
This error occurs because the camera calibration is missing.
Please refer to [camera_calibration](http://wiki.ros.org/camera_calibration) and follow the instruction.

idrobotiskidslab, 2018-07-06:
Thank you, I will try it later.
Hopefully can be solved. 
I will report to you.

closed 2018-07-08 by robotpilot