ROBOTIS-GIT/ROBOTIS-OP3 #48: Controlling Darwin with a Joystick

fvega99, 2018-07-18:
Hello,

I am trying to control Darwin Op2 with a joystick and I am having trouble figuring out how to read the input data from the controller. I currently have the controller connected to the computer through an application called jstest-gtk, yet I am not sure how to convert these values into if statements in order to control darwin.

ROBOTIS-zerom, 2018-07-23:
I don't know exactly what you want. Please write your questions in more detail.

fvega99, 2018-07-23:
i want to use the joy_node publisher in order to send data to a topic in
which the darwin op2 demo program will subscribe to in order to move
certain dynamixels. Yet, I am also having trouble installilng ros as it
states that it can not find the darwin_a library even though I have it
copied into the modules folder.

ROBOTIS-zerom, 2018-07-24:
Please refer to the link below. I hope it helps.

http://emanual.robotis.com/docs/en/platform/op2/ros_package/#robotis-op2-ros-packages

fvega99, 2018-07-24:
Thank you. 

Another question I have is how to move each individual actuator on Darwin a certain position based on the the joystick input .

ROBOTIS-zerom, 2018-07-24:
Please refer to the link below: 
http://emanual.robotis.com/docs/en/platform/op3/tutorials/#read-write-tutorial 

this is a description of OP3, but you can implement it by reference.

fvega99, 2018-07-24:
> tutorials/#read-write-tutorial

is it possible to do this without ros?

ROBOTIS-zerom, 2018-07-24:
If you want to do this without ROS, refer to the link below:
http://emanual.robotis.com/docs/en/platform/op/development/#linux-tutorials

fvega99, 2018-07-24:
> development/#linux-tutorials

I get the robot to read the present position, yet i am confused on how to
move the actuator forward to a desired position (similar to the action
editor in which the arm raises slightly).

robotpilot, 2018-08-08:
@ROBOTIS-zerom, @fvega99, Is this an end issue or ongoing issue?