ROBOTIS-GIT/ROBOTIS-OP3 #50: error when try op3_manager with XM540-W270-R

msaufyrohmad, 2018-07-20:
hi 
i try op3 manager with set of 15 xm540 servos. 

i already change the .robot file and add XM-540.device file. 

but when i try to run the op3manager with op3_manager.launch, this message appear. 

process[op3_localization-2]: started with pid [26270]
[ERROR] [1532100652.615660890]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.651648647]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.687631622]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.723618992]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.759606870]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.895751751]: Fail to control LED [[RxPacketError] Unknown error code!]
/dev/ttyACM0 added. (baudrate: 57600)
(/dev/ttyACM0) [ID:  1]         XM-540 added. 
(/dev/ttyACM0) [ID:  2]         XM-540 added. 
(/dev/ttyACM0) [ID:  3]         XM-540 added. 
(/dev/ttyACM0) [ID:  4]         XM-540 added. 
(/dev/ttyACM0) [ID:  5]         XM-540 added. 
(/dev/ttyACM0) [ID:  6]         XM-540 added. 
(/dev/ttyACM0) [ID:  7]         XM-540 added. 
(/dev/ttyACM0) [ID:  8]         XM-540 added. 
(/dev/ttyACM0) [ID:  9]         XM-540 added. 
(/dev/ttyACM0) [ID: 10]         XM-540 added. 
(/dev/ttyACM0) [ID: 11]         XM-540 added. 
(/dev/ttyACM0) [ID: 12]         XM-540 added. 
(/dev/ttyACM0) [ID: 13]         XM-540 added. 
(/dev/ttyACM0) [ID: 14]         XM-540 added. 
(/dev/ttyACM0) [ID: 15]         XM-540 added. 
(/dev/ttyACM0) [ID:200]        OPEN-CR added. 
[ INFO] [1532100654.028718785]: Load offsets...
[ WARN] [1532100654.348599766]: Head control - angle unit : 30
[ERROR] [1532100655.314481806]: [RobotisController] first bulk read fail!!
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid arguu ment
[adam_manager-1] process has died [pid 26241, exit code -6, cmd /home/husein/catkin_ws/devel/lib/adam_manager/adam_manager __name:=adam_manager __log:=/home/husein/.ros/log/7ff5bac6-8c2e-11e8-a852-240a64b1d379/adam_manager-1.log].
log file: /home/husein/.ros/log/7ff5bac6-8c2e-11e8-a852-240a64b1d379/adam_manager-1*.log

I think i already change what suppose to change, but i study here.

thanks You very much

saufy

ROBOTIS-Kayman, 2018-07-22:
Hello.

```
[ERROR] [1532100652.615660890]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.651648647]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.687631622]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.723618992]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.759606870]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1532100652.895751751]: Fail to control LED [[RxPacketError] Unknown error code!]
```
Upper Error means that you don't use OPEN-CR whose ID for communication is 200.
Please refer the [link](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#op3).

```
[ERROR] [1532100655.314481806]: [RobotisController] first bulk read fail!!
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injectorboost::lock_error >'
```
And second one has 2 solution.
1. Get superuser privilege while maintaining the bash environment please.
    ```
    $ sudo bash
    ```

2. Follow the `Other settings` of the [PC Setting](http://emanual.robotis.com/docs/en/platform/op3/recovery/#pc-setting) please.

ROBOTIS-Kayman, 2018-07-27:
If you have an another plan for power control not using OpenCR, you don't have to use OpenCR. UsbToDxnamixel is enough.
In the case of ROBOTIS-OP3, we used OpenCR in order to control the power source.
It's a [system block diagram](http://emanual.robotis.com/docs/en/platform/op3/quick_start/#system-block-diagram) of ROBOTIS-OP3.

200 is a default ID of OpenCR in the firmware for ROBOTIS-OP3.

ROBOTIS-Kayman, 2018-07-30:
I see. I finally know what is different between you and I.  
We used OpenCR like a dynamixel under the [U2D2](http://emanual.robotis.com/docs/en/parts/interface/u2d2/) in ROBOTIS-OP3.   
`/dev/ttyUSB0` is a portname of U2D2. This is an electronic structure of ROBOTIS-OP3.  
```
PC - U2D2 - OpenCR (200)
          - Dynamixel (1)
          - Dynamixel (2)
          - ...
```

Maybe it will work if you change the robot file and a launch file. The developer of OpenCr is in vacation until next week.  
Could you test following method.  
1. change a robot file
   - open a `OP3.robot` file.
   - replace `/dev/ttyUSB0` to `/dev/ttyACM0`.
2. change a launch file
   - open a `op3_manager.launch` file.
   - replace `/dev/ttyUSB0` to `/dev/ttyACM0`.

When my colleague is coming next week ,I'll ask him that whether it is possible to communicate with OpenCR and dynamixels using micro usb port of OpenCR under the firmware for ROBOTIS-OP3.

msaufyrohmad, 2018-08-01:
oh. we are connecting direct from PC --> OpenCR --> XM540-W270

I already change the .robot file, offset.yaml and dxl_init file. I change
to /dev/ttyACM1 for all joint in .robot file. i also change in launch file
to /dev/ttyACM1. the devices file also i use XM540-W270. After i run
roslaunch this output appear. help me please.  thank you very much.

process[op3_localization-2]: started with pid [28889]
[ INFO] [1533087892.716516643]: manager->init
[ERROR] [1533087892.774784135]: Torque on DXLs! [[RxPacketError] Unknown
error code!]
[ERROR] [1533087892.810790056]: Torque on DXLs! [[RxPacketError] Unknown
error code!]
[ERROR] [1533087892.846772185]: Torque on DXLs! [[RxPacketError] Unknown
error code!]
[ERROR] [1533087892.882775876]: Torque on DXLs! [[RxPacketError] Unknown
error code!]
[ERROR] [1533087892.918777710]: Torque on DXLs! [[RxPacketError] Unknown
error code!]
[ERROR] [1533087893.054881841]: Fail to control LED [[RxPacketError]
Unknown error code!]
/dev/ttyACM1 added. (baudrate: 57600)
(/dev/ttyACM1) [ID:  1]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  2]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  3]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  4]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  5]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  6]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  7]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  8]    XM-540-W270 added.
(/dev/ttyACM1) [ID:  9]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 10]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 11]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 12]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 13]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 14]    XM-540-W270 added.
(/dev/ttyACM1) [ID: 15]    XM-540-W270 added.
(/dev/ttyACM1) [ID:200]        OPEN-CR added.
[ERROR] [1533087893.244897528]: JOINT[head_yaw] does NOT respond!!
[ERROR] [1533087893.326990026]: JOINT[r_ank_pitch] does NOT respond!!
[ERROR] [1533087893.409011186]: JOINT[r_ank_pitchS] does NOT respond!!
[ERROR] [1533087893.491112007]: JOINT[r_ank_roll] does NOT respond!!
[ERROR] [1533087893.573152217]: JOINT[r_ank_rollS] does NOT respond!!

-saufy

msaufyrohmad, 2018-08-01:
i dont have U2D2 but i just try to use USB2Dynamixel and connect RS485 in
USB2Dynamixel to OpenCR.

i have 2 questions:
1. do i need to download opencr_op3 firmware into opencr before i connect
opencr to USB2Dynamixel?
2. do i need to download anything into USB2Dynamixel? or i need to download
any firmware first?

thank you very much

saufy

msaufyrohmad, 2018-08-01:
> Yes, you can use USB2Dynamixel instead of U2D2.
>
> Q1 : If you want to use function of power control or sensor(gyro,
> accelermeter), you have to download a firmware for ROBOTIS-OP3 into OpenCR.
> (emanual
> <http://emanual.robotis.com/docs/en/parts/controller/opencr10/#op3>)
>
> Q2 : you don't have to do anything.
>
> Good luck.

thank you, it progressing.

i already can set the torque. but the ping still not work

[ INFO] [1533097846.650661842]: Torque on DXLs!
/dev/ttyUSB0 added. (baudrate: 2000000)
(/dev/ttyUSB0) [ID:  1]    XM-540-W270 added.
...
...
(/dev/ttyUSB0) [ID: 15]    XM-540-W270 added.
(/dev/ttyUSB0) [ID:200]        OPEN-CR added.
[ERROR] [1533097213.167007397]: ADAM JOINT[head_yaw] does NOT respond!!
[ERROR] [1533097213.245370361]: ADAM JOINT[r_ank_pitch] does NOT respond!!
[ERROR] [1533097213.323732783]: ADAM JOINT[r_ank_pitchS] does NOT respond!!

this is my connection

PC USB Port ---> USB2Dynamixel (/dev/ttyUSB0) ---> OpenCR (load with
opencr_op3 firmware) --> XM540-W270 (15 units in series).

can u help me?

saufy

ROBOTIS-Kayman, 2018-08-01:
I don't know what is problem.

Could you give me a photo of your hardware configuration? I want to see it.
And please give me your robot file too.

msaufyrohmad, 2018-08-01:
ok this is my adam.robot file. below is image,

[ control info ]
control_cycle = 8   # milliseconds

[ port info ]
# PORT NAME  | BAUDRATE  | DEFAULT JOINT
/dev/ttyUSB0 | 2000000  | r_ank_pitch

[ device info ]
# TYPE    | PORT NAME    | ID  | MODEL          | PROTOCOL | DEV NAME
 | BULK READ ITEMS
dynamixel | /dev/ttyUSB0 | 1   | XM540-W270     | 2.0      | r_ank_pitch
| present_position
dynamixel | /dev/ttyUSB0 | 2   | XM540-W270     | 2.0      | r_ank_pitchS
 | present_position
dynamixel | /dev/ttyUSB0 | 3   | XM540-W270     | 2.0      | r_ank_roll
 | present_position
dynamixel | /dev/ttyUSB0 | 4   | XM540-W270     | 2.0      | r_ank_rollS
| present_position
dynamixel | /dev/ttyUSB0 | 5   | XM540-W270     | 2.0      | r_knee_roll
| present_position
dynamixel | /dev/ttyUSB0 | 6   | XM540-W270     | 2.0      | r_knee_rollS
 | present_position
dynamixel | /dev/ttyUSB0 | 7   | XM540-W270     | 2.0      | r_knee_rollS2
| present_position
dynamixel | /dev/ttyUSB0 | 8   | XM540-W270     | 2.0      | r_knee_rollS3
| present_position
dynamixel | /dev/ttyUSB0 | 9   | XM540-W270     | 2.0      | r_hip_roll
 | present_position
dynamixel | /dev/ttyUSB0 | 10   | XM540-W270    | 2.0      | r_hip_rollS
| present_position
dynamixel | /dev/ttyUSB0 | 11   | XM540-W270    | 2.0      | r_hip_pitch
| present_position
dynamixel | /dev/ttyUSB0 | 12   | XM540-W270    | 2.0      | r_hip_pitchS
 | present_position
dynamixel | /dev/ttyUSB0 | 13   | XM540-W270    | 2.0      | r_hip_yaw
| present_position
dynamixel | /dev/ttyUSB0 | 14   | XM540-W270    | 2.0      | r_hip_yawS
 | present_position
dynamixel | /dev/ttyUSB0 | 15   | XM540-W270    | 2.0      | head_yaw     |
present_position
sensor    | /dev/ttyUSB0 | 200 | OPEN-CR        | 2.0      | open-cr
| button, present_voltage, gyro_x, gyro_y, gyro_z, acc_x, acc_y, acc_z,
roll, pitch, yaw
~

The image: USB2Dynamixel connected to laptop USB /dev/ttyUSB0 and RS485 to
OpenCR. OpenCR connect to XM540-W270 in series.

thnks,

saufy

ROBOTIS-Kayman, 2018-08-01:
I think there is no problem in the robot file.

How about checking ID, baud rate, protocol version of dynamixels?
They have to be same between a robot file and dynamixels.

I'll test it in the similar condition and let you know the result.

msaufyrohmad, 2018-08-01:
> It works for me.
> This is my test condition.
>
>    - dynamixels
>       - XM540-W150
>          - ID : 1
>          - baudrate : 2000000
>       - XM540-W270
>          - ID : 2
>          - baudrate : 2000000
>       - robot file
>
> [ control info ]
> control_cycle = 8   # milliseconds
>
> [ port info ]
> # PORT NAME  | BAUDRATE  | DEFAULT JOINT
> /dev/ttyUSB0 | 2000000   | r_sho_pitch
>
> [ device info ]
> # TYPE    | PORT NAME    | ID  | MODEL          | PROTOCOL | DEV NAME       | BULK READ ITEMS
> dynamixel | /dev/ttyUSB0 | 1   | XM540-W150     | 2.0      | r_sho_pitch    | present_position, position_p_gain, position_d_gain, position_i_gain
> dynamixel | /dev/ttyUSB0 | 2   | XM540-W270     | 2.0      | l_sho_pitch    | present_position, position_p_gain, position_d_gain, position_i_gain
> sensor    | /dev/ttyUSB0 | 200 | OPEN-CR        | 2.0      | open-cr        | button, present_voltage, gyro_x, gyro_y, gyro_z, acc_x, acc_y, acc_z, roll, pitch, yaw
>
>
>
>    - execution screen
>    [image: image]
>    <https://user-images.githubusercontent.com/16040148/43506903-947f8f18-95a6-11e8-9314-eeee60205da6.png>

i still cant. ok i tell u my configuration

Hardware Config (image attached)
PC USB ----rs485----> Dynamixel --> .... Dynamixel ---,........(15 units)
-----rs485---> OpenCR --- power cable ---> plug

Software Config
baudrate 2000000
port /dev/ttyUSB0

adam.robot configured like urs

but, i have 7 master and 8 slaves servos. thus this that cause the problem?

do i need to put all servos (including the slaves) in the .robot file or
just put the master? for now i put all master and slaves in .robot file

thank you for your patient helping me out

saufy

ROBOTIS-Kayman, 2018-08-02:
What kind of method did you use for master and slave?
Is it using [direct cable](http://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#drive-mode10) or [secondary id](http://emanual.robotis.com/docs/en/dxl/x/xm540-w270/#secondaryshadow-id12)?

msaufyrohmad, 2018-08-02:
direct cable. some is cross and some is straight. depend on the design (for
cw or ccw slaves). already tested and works fine

saufy

ROBOTIS-Kayman, 2018-08-02:
In my case, both of putting all servos to robot file and putting only masters are working.

I am wondering why number of slave is bigger than number of master. 
As I know, master and slave are connected one by one.

- robot file
```
[ control info ]
control_cycle = 8   # milliseconds

[ port info ]
# PORT NAME  | BAUDRATE  | DEFAULT JOINT
/dev/ttyUSB0 | 2000000   | r_sho_pitch

[ device info ]
# TYPE    | PORT NAME    | ID  | MODEL          | PROTOCOL | DEV NAME       | BULK READ ITEMS
dynamixel | /dev/ttyUSB0 | 1   | XM540-W150     | 2.0      | r_sho_pitch    | present_position, position_p_gain, position_d_gain, position_i_gain
dynamixel | /dev/ttyUSB0 | 2   | XM540-W270     | 2.0      | l_sho_pitch    | present_position, position_p_gain, position_d_gain, position_i_gain
dynamixel | /dev/ttyUSB0 | 3   | XM540-W270     | 2.0      | l_sho_pitchS   | present_position, position_p_gain, position_d_gain, position_i_gain
sensor    | /dev/ttyUSB0 | 200 | OPEN-CR        | 2.0      | open-cr        | button, present_voltage, gyro_x, gyro_y, gyro_z, acc_x, acc_y, acc_z, roll, pitch, yaw
```
- HW configuration
[image]

msaufyrohmad, 2018-08-03:
i think i have some package that not installed properly. because i keep
getting the same error message. i am using ubuntu 16.04, not linux mint,
its have some problem here?

(/dev/ttyUSB0) [ID: 15]    XM-540-W270 added.
(/dev/ttyUSB0) [ID:200]        OPEN-CR added.
[ERROR] [1533261427.075738545]: ADAM JOINT[head_yaw] does NOT respond!!
[ERROR] [1533261427.154023274]: ADAM JOINT[r_ank_pitch] does NOT respond!!
[ERROR] [1533261427.232379208]: ADAM JOINT[r_ank_pitchS] does NOT respond!!

more than one slave is needed for our knee. one as master, one clock wise
slave and 2 counter clock wise slave. but i think this is not the problem
here.

thank you very much for following my problem one by one

saufy

ROBOTIS-Kayman, 2018-08-03:
Ubuntu 16.04 is fine as well. it also looks no problem there are one master and two slaves.

msaufyrohmad, 2018-08-03:
where is the log file that i can see the error?
i want to see where is the missing link between ROS and dxl motor

thanks

saufy

ROBOTIS-Kayman, 2018-08-06:
There is no log file.
But If you change the flag for debug mode in [robotis_controller.cpp](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/cd58b92487b701380ff606c6f430e891f1c12ad2/robotis_controller/src/robotis_controller/robotis_controller.cpp#L38), you can see the more log in the screen.

msaufyrohmad, 2018-08-06:
ok thank u. ill try

saufy

robotpilot, 2018-08-08:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue. Thanks.

closed 2018-08-08 by robotpilot

ROBOTIS-Kayman, 2018-08-20:
Can I see the picture of your hardware?
I don't know what thr problem is for now.

msaufyrohmad, 2018-08-23:
this is my hardware configuration

PC USB Port --> usb2dynamixel --> RS485 to OpenCR -> RS485 OpenCR -> 3
XM540-W270 in series

i used op3_manager package with below adam.robot file

[ control info ]
control_cycle = 8   # milliseconds

[ port info ]
# PORT NAME  | BAUDRATE  | DEFAULT JOINT
/dev/ttyUSB0 | 2000000  | head_yaw

[ device info ]
# TYPE    | PORT NAME    | ID  | MODEL          | PROTOCOL | DEV NAME
 | BULK READ ITEMS
dynamixel | /dev/ttyUSB0 | 1   | XM-540-W270    | 2.0      | r_ank_pitch
| present_position
dynamixel | /dev/ttyUSB0 | 2   | XM-540-W270    | 2.0      | r_ank_pitchS
 | present_position
dynamixel | /dev/ttyUSB0 | 3   | XM-540-W270    | 2.0      | r_ank_roll
 | present_position
dynamixel | /dev/ttyUSB0 | 4   | XM-540-W270    | 2.0      | r_ank_rollS
| present_position
dynamixel | /dev/ttyUSB0 | 5   | XM-540-W270    | 2.0      | r_knee_roll
| present_position
dynamixel | /dev/ttyUSB0 | 6   | XM-540-W270    | 2.0      | r_knee_rollS
 | present_position
dynamixel | /dev/ttyUSB0 | 7   | XM-540-W270    | 2.0      | r_knee_rollS2
| present_position
dynamixel | /dev/ttyUSB0 | 8   | XM-540-W270    | 2.0      | r_knee_rollS3
| present_position
dynamixel | /dev/ttyUSB0 | 9   | XM-540-W270    | 2.0      | r_hip_roll
 | present_position
dynamixel | /dev/ttyUSB0 | 10   | XM-540-W270   | 2.0      | r_hip_rollS
| present_position
dynamixel | /dev/ttyUSB0 | 11   | XM-540-W270   | 2.0      | r_hip_pitch
| present_position
dynamixel | /dev/ttyUSB0 | 12   | XM-540-W270   | 2.0      | r_hip_pitchS
 | present_position
dynamixel | /dev/ttyUSB0 | 13   | XM-540-W270   | 2.0      | r_hip_yaw
| present_position
dynamixel | /dev/ttyUSB0 | 14   | XM-540-W270   | 2.0      | r_hip_yawS
 | present_position
dynamixel | /dev/ttyUSB0 | 15   | XM-540-W270   | 2.0      | head_yaw
 | present_position
sensor    | /dev/ttyUSB0 | 200 | OPEN-CR        | 2.0      | open-cr
|
button,present_voltage,gyro_x,gyro_y,gyro_z,acc_x,acc_y,acc_z,roll,pitch,yaw
~

below .launch file

   <param name="offset_file_path"         value="$(find
adam_manager)/config/offset.yaml"/>
    <param name="robot_file_path"          value="$(find
adam_manager)/config/adam.robot"/>
    <param name="init_file_path"           value="$(find
adam_manager)/config/dxl_init_adam.yaml"/>
    <param name="device_name"              value="/dev/ttyUSB0"/>

    
    <node pkg="adam_manager" type="adam_manager" name="adam_manager"
output="screen">
         <param name="angle_unit" value="30" />
        </node>

the baudrate is 2000000.

everything else is following op3_manager package. aid got below output.

[ INFO] [1534988445.602163817]: Torque on DXLs!
[ INFO] [1534988445.714108317]: LED OK

/dev/ttyUSB0 added. (baudrate: 2000000)
ADAM (/dev/ttyUSB0) [ID:  1]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  2]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  3]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  4]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  5]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  6]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  7]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  8]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID:  9]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 10]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 11]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 12]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 13]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 14]    XM-540-W270 added.
ADAM (/dev/ttyUSB0) [ID: 15]    XM-540-W270 added.
(/dev/ttyUSB0) [ID:200]        OPEN-CR added.
[ERROR] [1534988445.947550839]: ADAM JOINT[head_yaw] does NOT respond!!
[ERROR] [1534988446.025891266]: ADAM JOINT[r_ank_pitch] does NOT respond!!
[ERROR] [1534988446.104184619]: ADAM JOINT[r_ank_pitchS] does NOT respond!!

i dont know what is the error and i really hope get helps from robotis
engineers.

thank  you very much

saufy rohmad
robopreneur sdn bhd

ROBOTIS-Kayman, 2018-08-24:
Did LED of Dynamixels blink when you run the manager.launch file?
Otherwise, power was not supplied to the Dynamixel.

And I think your hardware configuration is fine. You have to debug using modifing the source code. (ex. [robotis_controller](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/cd58b92487b701380ff606c6f430e891f1c12ad2/robotis_controller/src/robotis_controller/robotis_controller.cpp#L326))

msaufyrohmad, 2018-08-24:
No LED blink in the servors.

but the torque enabling is worked fine.

[ INFO] [1535101606.623536048]: Torque on DXLs!

(/dev/ttyUSB0) [ID:200]        OPEN-CR added.

PING Packet Handler id:15 result:-3001 // this is comment inside packet
handler.c in dynamixel sdk
PING Packet Handler id:15 result:-3001
[ERROR] [1535101606.844833668]: ADAM JOINT[head_yaw] does NOT respond!!  //
this is commend inside robotis_controller that u mention above
PING Packet Handler id:1 result:-3001

since the torque enable is working, i think it is not about the power
problem. thank you. i really need help on this

saufy

ROBOTIS-Kayman, 2018-08-28:
When the power is on, the servo LED blinks once. Will you check when it blinks?
In normal condition, LED blinks once when you run the manager.launch file.
At the beginning of the manager, we send a packet to OpenCR to turn on dynamixel. Then the LED will blink. If it does not blink then OpenCR seems to have a problem.

And I want to see the picture of your robot.
Using text alone will not help you anymore.

msaufyrohmad, 2018-08-30:
When the power is on, the servo LED blinks once. Will you check when it
blinks?

>> No, the servos not blink at all

In normal condition, LED blinks once when you run the manager.launch file.
>> No all servos LED  not blinking

At the beginning of the manager, we send a packet to OpenCR to turn on
dynamixel. Then the LED will blink. If it does not blink then OpenCR seems
to have a problem.

>> not blink but openCR is working fine with other codes.

And I want to see the picture of your robot.
[image: op3_manager.jpeg]

the hardware configuration is following what op3 is doing. we dont have a
robot yet. just series of servos.

saufy, thank you very much

ROBOTIS-Kayman, 2018-09-03:
When the power is on, the servo LED blinks once. Will you check when it
blinks?
> No, the servos not blink at all
>> It's impossible. AN XM540's LED also blinks when power is supplied. I think that the problem is about power.

The picture of series of servos is fine. Could you upload the image to this issue or send me via e-mail?
I can't see the images you uploaded before.

luqmansen, 2020-09-10:
> If you have another plan for power control and sensor not using OpenCR, you don't have to use OpenCR. UsbToDxnamixel is enough.
> In the case of ROBOTIS-OP3, we used OpenCR in order to control the power source.
> It's a [system block diagram](http://emanual.robotis.com/docs/en/platform/op3/quick_start/#system-block-diagram) of ROBOTIS-OP3. and please refer to the [control table](http://emanual.robotis.com/docs/en/platform/op3/quick_start/#ram-area) of OpenCR.

Hello, regarding this statement, I want to clarify that I could totally not using an OpenCR to control a daisy-chained N number of dynamixel.

Currently, this is my configuration 
PC --- U2D2 ----  MX-64 --- MX-64 --.... (24 * MX-64)
Based on [this](https://emanual.robotis.com/docs/en/parts/interface/u2d2/#pc-to-dynamixel) reference this should be totally supported. I'm only using the data pin from the U2D2 and then the power is using from another power source. My concern is about the bulk R/W performance when used in the realtime application (mine was a bipedal robot). 

Another issue that I want to ask when  trying to using the op3_manager code example, I experienced this problem 
>[ERROR] [1532100652.615660890]: Torque on DXLs! [[RxPacketError] Unknown error code!]

and your solution was 

> Upper Error means that you don't use OPEN-CR whose ID for communication is 200.

I'm unable to find the communication id of U2D2 anywhere on this [docs](https://emanual.robotis.com/docs/en/parts/interface/u2d2/)

Thank you in advance

luqmansen, 2020-09-29:
hello @ROBOTIS-Kayman, currenty I'm using this configuration

PC --- OpenCR ---- MX-64 --- MX-64 --.... (24 * MX-64)

and my problem still persists like my previous question, could you help me regarding my problem? I'm stuck for quite a while now. Thank you in advance

ROBOTIS-Will, 2020-10-05:
Hi @luqmansen ,
I'm sorry about the late reply.

In case of OP3, the controller reads the Present Position of 20 DYNAMIXEL with 8ms control frequency at 2Mbps so you should be fine with 24 DYNAMIXEL. (refer to : https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/OP3.robot)

OP3 uses the OpenCR for IMU and power supply, therefore, using U2D2 instead of OpenCR won't need the power on sequence which causes the aforementioned error in your case. 
If you are going to use the OpenCR, you should use a proper OpenCR firmware for OP3, or else, you can use U2D2 and supply power to DYNAMIXEL with a separate power line. 
In the latter case using U2D2, you may be able to avoid the error by commenting out the OpenCR related code [line 170~207 of op3_manager.cpp](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/src/op3_manager.cpp#L170) file and [line 239](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/src/op3_manager.cpp#L239) which adds IMU sensor on the OpenCR.

Thank you.