ROBOTIS-GIT/ROBOTIS-OP3 #51: Connection between OpenCR and Intel NUC

Purmpol, 2018-09-15:
Hi, 

I'm learning ROS and going to make my own custom robot. It may be OP3-like robot, based on OpenCR & XL-430 (for more economical).  Before I buy these items, I would like to know something.

Q: How OpenCR "physically" connect to Intel NUC on OP3?
The picture on e-manual, there's only one USB port for firmware download (not always connect to NUC), right?
[image]

ROBOTIS-Kayman, 2018-09-18:
Hi,

We used a [U2D2](http://emanual.robotis.com/docs/en/parts/interface/u2d2/) to connect between NUC and OP3.
[NUC] ---(USB)--- [U2D2] ---(RS485)--- [DXL and OpenCR]

If you make the own firmware for OpenCR, you don't have to connect like this.
you can connect the OpenCR directly (using micro USB port).

Purmpol, 2018-09-18:
Hi Kayman,

Yes. Understand.
I prefer the way of connection in OP3. I think it makes robot look compact and tidy by internal connection. There's no messy external wire.

Thanks again for your support.

closed 2018-09-21 by ROBOTIS-Kayman