ROBOTIS-GIT/ROBOTIS-OP3 #52: Wiring Manual for the OP3

ferri69, 2018-10-21:
Hello,
 
do you have an „Wiring Manual“ for the OP3.
I don´t find anything.
If you may have a few pictures for me, this would help me a lot.
(Open CR, Intel NUC....)
 
Thanks for your help

ROBOTIS-Kayman, 2018-10-24:
Hello, @ ferri69

Sorry for late reply.
There is no wiring documentation at this time.

We will soon make a manual and release it.
Please wait.

I attached some images for you.
[image]
[image]
[image]

ROBOTIS-Kayman, 2018-12-10:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2018-12-10 by ROBOTIS-Kayman