ROBOTIS-GIT/ROBOTIS-OP3 #54: Fail to get foot step from planner

ddliugit, 2018-12-04:
I am doing a simulation with op3. While I am using the foot planner, I met several problems. 

First it is the tf fixed frame in rviz. I fixed it by remapping body_link to world frame. Please fix that problem in the next update.
Second, the problem is caused by the foot step service. In the ui part, I think you didn't start the foot plan service. The log said the service is not called. 
I start the service manually. Please fix this problem in the next update.

The third problem I am still working on and waiting for your help is that 
Start pose is not accessible.

ROBOTIS-Kayman, 2018-12-05:
Hi

- Fisrt, Second one
  I think you did now follow the [tutorial of online walking](http://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-control-upgraded-walking-using-footstep-planner).
  Please see [the tutorial](http://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-control-upgraded-walking-using-footstep-planner) and follow it.  
  When I checked it, I found the missing part in `Run the program` section. 
  Please run the `op3_localization` manually.
  ```
  $ rosrun op3_localization op3_localization
  ```

- Third one
I don't understand `start pose`. Could you give me more information about that?

ddliugit, 2018-12-05:
Dear Kayman,

Yes, I followed this tutorial.

After running the op3_localization manually, it solved the first problem.
However, the second problem still exists. "cannot call services" shows up
after I click the plan button.

[image: image.png]

[image: image.png]

Can you help me fix this problem?

Best regards,
Jack

ddliugit, 2018-12-05:
> cannot call services means that footstep_planner does not be launched.
> It's one of humanoid_navigation packages.
> How about deleting the humanoid_navigation package and clone it from
> following link
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_ROBOTIS-2DGIT_humanoid-5Fnavigation&d=DwMFaQ&c=slrrB7dE8n7gBJbeO0g-IQ&r=hkuqmUNuMbV2eMHNPYrllQ&m=IVSW305dCd-v2hFvMxaAdy85thC-gF0XusqWDNABvbA&s=HuSbNWKyhuwyynaJ92YB6otpkY4zUDv_h25DD4SujxE&e=>,
> compile it(catkin_make).
>
> $ git clone https://github.com/ROBOTIS-GIT/humanoid_navigation.git

I downloaded it and recompiled it. It does not work.

problem 1: cannot call service. still exists

so I just run
$ roslaunch footstep_planner footstep_planner.launch

manually run this run(don't know if it is correct, because there are many
launch files in the footstep_planner package).

After running this command, problem1 has disappeared.

problem2 shows up.

in the command window which runs footstep_planner, the error message is
that "Distance map hasn't been initialized yet."

I have no idea about this problem and not sure if the first method is
correct. Can you help me please ?

regards,
Jack

ddliugit, 2018-12-05:
Dear Kayman,

Then I tried different way.

"Distance map hasn't been initialized yet."

may be caused by the map service which has been launched in the
footstep_planner.launch.

Then, I launched the complete version of this footstep_planner

$ roslaunch footstep_planner footstep_planner_complete.launch

This launch will open a rviz gui again which I closed it manually.

Then, I click the plan button in dynamixel gui.

Probem3 shows up.

In the command window which runs the footstep_planner_complete.launch, it
shows up:

ERROR] [1544013860.495347652, 909.499000000]: Start pose (0.000000 0.000000
0.000000) not accessible.

The error message in the gui is that "fail to get foot step from planner"

See the log for the details:

jack@jack:~/catkin_ws$ roslaunch footstep_planner footstep_planner.launch
> ... logging to
> /home/jack/.ros/log/a55a2902-f888-11e8-b81d-b06ebf5bcf8f/roslaunch-jack-16319.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
>
> started roslaunch server http://jack:45617/
>
> SUMMARY
> ========
>
> PARAMETERS
>  * /footstep_planner/accuracy/cell_size: 0.01
>  * /footstep_planner/accuracy/collision_check: 2
>  * /footstep_planner/accuracy/num_angle_bins: 64
>  * /footstep_planner/allocated_time: 7.0
>  * /footstep_planner/changed_cells_limit: 20000
>  * /footstep_planner/diff_angle_cost: 0.0
>  * /footstep_planner/foot/max/inverse/step/theta: -0.4
>  * /footstep_planner/foot/max/inverse/step/x: -0.1
>  * /footstep_planner/foot/max/inverse/step/y: 0.12
>  * /footstep_planner/foot/max/step/theta: 0.7
>  * /footstep_planner/foot/max/step/x: 0.22
>  * /footstep_planner/foot/max/step/y: 0.28
>  * /footstep_planner/foot/origin_shift/x: 0.02
>  * /footstep_planner/foot/origin_shift/y: 0.0
>  * /footstep_planner/foot/separation: 0.14
>  * /footstep_planner/foot/size/x: 0.16
>  * /footstep_planner/foot/size/y: 0.1
>  * /footstep_planner/foot/size/z: 0.015
>  * /footstep_planner/footsteps/theta: [0.0, 0.0, 0.0, 0...
>  * /footstep_planner/footsteps/x: [0.0, 0.22, 0.0, ...
>  * /footstep_planner/footsteps/y: [0.14, 0.14, 0.26...
>  * /footstep_planner/forward_search: False
>  * /footstep_planner/heuristic_type: PathCostHeuristic
>  * /footstep_planner/initial_epsilon: 8.0
>  * /footstep_planner/max_hash_size: 65536
>  * /footstep_planner/planner_type: ARAPlanner
>  * /footstep_planner/search_until_first_solution: False
>  * /footstep_planner/step_cost: 0.1
[... 69 lines cut ...]
>  * /footstep_planner/step_range/y: [0.28, 0.12, 0.12...
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>
> NODES
>   /
>     footstep_planner (footstep_planner/footstep_planner_node)
>     map_server (map_server/map_server)
>     rviz (rviz/rviz)
>
> ROS_MASTER_URI=http://localhost:11311
>
> process[map_server-1]: started with pid [16908]
> process[rviz-2]: started with pid [16909]
> process[footstep_planner-3]: started with pid [16917]
> [rviz-2] process has finished cleanly
> log file:
> /home/jack/.ros/log/a55a2902-f888-11e8-b81d-b06ebf5bcf8f/rviz-2*.log
> [ERROR] [1544013860.495347652, 909.499000000]: Start pose (0.000000
> 0.000000 0.000000) not accessible.
>
> Can you help me figure this error out?

Best regards,
Jack

ROBOTIS-Kayman, 2018-12-06:
Is it right that you ran `$ roslaunch op3_gui_demo op3_demo_walking.launch`?

If you run the `op3_gui_demo/launch/op3_demo_walking.launch`, `op3_navigation/launch/op3_footstep_planner.launch` is run and `footstep_planner` is run also.

In your case, you ran `footstep_planner` manually except proper parameters. I think it's why `footstep_planner` did not work.

Please refer to [launch file of  `op3_navigation` package](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Tools/tree/master/op3_navigation/launch).

ddliugit, 2018-12-07:
Thank you very much. I applied the above method and it works.

I am working on the project with foot planning in uneven ground and 6D
navigation. Is there a package in op3? If yes, how to change the file with
less time?

I try to use a vigil foot step planning in ros and it seems very hard to
change. I don't know where should I change if I still want to use the
Dynamixel guy like 2d foot planning for the online walking.

Can you give me some guidance and tutorial about changing file in op3
package?

Best regards,
Jack

ddliugit, 2018-12-07:
Thank you very much. I applied the above method and it works.

I am working on the project with foot planning in uneven ground and 6D
navigation. Is there a package in op3? If yes, how to change the file with
less time?

I try to use a vigir  foot step planning in ros and it seems very hard to
change. I don't know where should I change if I still want to use the
Dynamixel GUI like 2d foot planning for the online walking.

Can you give me some guidance and tutorial about changing file in op3
package?

Best regards,
Jack

ROBOTIS-Kayman, 2018-12-10:
We don't have a footstep planner for 6D in ROBOTIS-OP3.
And I have never used a vigir footstep planner.
I do not know if I can help you.

To use a GUI like OP3, you need to create it manually. It's created using an interactive marker.
Please refer to [the source code of the marker part](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Tools/blob/master/op3_gui_demo/src/qnode_preview_walking.cpp) and [the wiki](http://wiki.ros.org/interactive_markers).

ddliugit, 2018-12-10:
Thanks, Kayman.

You help me a lot.

ROBOTIS-Kayman, 2018-12-10:
I wish you luck.

Best regards, 
Kayman

closed 2018-12-10 by ROBOTIS-Kayman