ROBOTIS-GIT/ROBOTIS-OP3 #58: Disable smoothing in direct_control_module

VasilyevGeorgy, 2018-12-17:
Hi!
Thanks for your previous tip, however a new problem has appeared.
In my walking algorithm I generate joint servos angles with quite little shift and then publish them in corresponding topic.
But when I try to do this in direct control mode there presents some smoothing that  disturb correct legs swinging trajectories.
May I somehow disable this smoothing effect?

Best regards,
George

ROBOTIS-Kayman, 2018-12-18:
Hi again,

What kind of method did you use?
I recommanded you to use the first one (method to use framework : robotis_controller).
The second method to use `direct_control_module` make a trajectory in the module for smothing.

Could you try the first method?
- Set module to none
  - make a message and put data to `none`
  - send it to `/robotis/enable_ctrl_module` 
- Control joint angles
  - make and send a topic to `/robotis /set_joint_states`

Regards,
Kayman

ROBOTIS-Kayman, 2018-12-19:
I understood your situation. 
I will inform the contents to the developer in charge. 
For a while I think you should use `direct_control_module` for simulation.
I'm sorry for your inconvinience.

Regards, 
Kayman

VasilyevGeorgy, 2018-12-19:
Thank you very much!

Best regards,
George

ROBOTIS-Kayman, 2018-12-20:
Hi,

The Problem is fixed. (https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/pull/72)
Could you test it in your simulation?

Regards, 
Kayman

VasilyevGeorgy, 2019-01-28:
Sorry for the late answer
Thanks, now I can change servos position in Gazebo
But what about 'velocity' parameter?
Every time I try to move servo, the rotation has the same velocity even it has been changed
How can I control servos' velocities directly?

Regards,
George

ROBOTIS-Kayman, 2019-01-31:
Sorry, George.
We do not support changing the 'velocity' parameter. 
How about changing the source code?
This is the path of [robotis_controller](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/master/robotis_controller/src/robotis_controller/robotis_controller.cpp).

Regards,
Kayman

VasilyevGeorgy, 2019-01-31:
Ok, thanks for your tip!

Regards,
George

closed 2019-02-27 by gpwhd07