ROBOTIS-GIT/ROBOTIS-OP3 #60: Unable to launch op3_manager on ROBOTIS OP3

VasilyevGeorgy, 2019-03-14:
After OP3's packages upgrade the robot doesn't take initial position.
If I try to launch op3_manager in console there appears an error: 
```
Unable to launch [op3_manager-2]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log file
```

Regards
George

P.S.: buttons on back also stopped to work.

ROBOTIS-Kayman, 2019-03-15:
Hi

Could you tell me how to upgrade to OP3
and tell me more about the error message?

Regards,
Kayman

VasilyevGeorgy, 2019-03-15:
OP3 terminates with an error
GDB log:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range

Thread 13 "op3_manager" received signal SIGABRT, Aborted.
```

VasilyevGeorgy, 2019-03-15:
Kinetic sync seemed to have introduced an ABI break. A complete, clean rebuild of the workspace was required, not just compiling again. Now it works fine

ROBOTIS-Kayman, 2019-03-18:
I am glad to be well.
If you have any other questions, please open the issue.
Thanks

closed 2019-03-18 by ROBOTIS-Kayman