ROBOTIS-GIT/ROBOTIS-OP3 #61: issue on direct control mode

yuan0623, 2019-03-16:
Hi, Our lab just purchased a Darwin-OP3. However, I have some issues with direct control mode. Here is the video [link:](https://photos.app.goo.gl/ErftgxeJbRwY678D7). From this video, I have two issues.

1. From the set joint position control, I only have 'r_sho_pitch', 'l_sho_pitch' changes with time, Other joints should be fixed. But the from the video, you can clearly tell that the joints on the leg moves along the time as well, and so does the 'x_sho_roll'! That's weird.

2. In the demo walking mode, the joints' behaviour are really impressive, they respond fast and agile. However, when I publish the reference position to 'robotis/direct_control/set_joint_state' in direct control mode, the joints move slow. When I check the joint gain, I think they are large enough (Kp = 800, Ki=0 , Kd=0). I'm thinking maybe there is a safety reason here? I know that the motor is totally capable of fast response. Or I need to directly talking to the motor without ros-manager?
The code I use corresponding to the video is here:
```python
#!/usr/bin/env python
import rospy
from sensor_msgs.msg import JointState
import numpy as np
def pub_head_pan():
    pub = rospy.Publisher('/robotis/direct_control/set_joint_states',JointState,queue_size = 30)
    rospy.init_node('test_joint_node')
    rate = rospy.Rate(50)
    count = 0
    jointstate = JointState()
    jointstate.position = np.zeros(20)
    while not rospy.is_shutdown():
        count = count+1
        print(count)
        desired_position = np.sin(0.01*count)
        jointstate.name = ['r_sho_pitch','l_sho_pitch','r_sho_roll',
        'l_sho_roll','r_el','l_el','r_hip_yaw','l_hip_yaw',
        'r_hip_roll','l_hip_roll','r_hip_pitch','l_hip_pitch',
        'r_knee','l_knee','r_ank_pitch','l_ank_pitch',
        'r_ank_roll','l_ank_roll','head_pan','head_tilt']

        jointstate.position[0]=np.sin(0.02*count)
        jointstate.position[1]=np.cos(0.02*count)
        #jointstate.position[19]=np.cos(0.02*count)
        jointstate.position[10] = 0.505/2
        jointstate.position[11] = -0.505/2
        jointstate.position[12] = -0.505
        jointstate.position[13] = 0.505
        jointstate.position[14] = -0.505/2
        jointstate.position[15] = 0.505/2
        jointstate.velocity = []
        jointstate.effort = []
        pub.publish(jointstate)
        rate.sleep()
if __name__ == '__main__':
    try:
        pub_head_pan()
    except rospy.ROSInterruptException:
        pass
```
I need to implement our own online walking algorithm, so I need to make sure that the joints of the robot is capable of fast response.
A sample code of yours is going to be really helpful!!!
Hope you can help me.

Best

Yuan

ROBOTIS-Kayman, 2019-03-18:
Hi~

1. I'll check and answer to you later.

2. I think `direct_control module` is not suitable for you. when an user sends the target position of joints, `direct_control module` make its own trajectory using theminimum jerk. It's why it moves slow.
I recommand you to use direct mode of `robotis_controller`. 

- Set module to none
  - make a message and put data to `none`
  - send it to `/robotis/enable_ctrl_module` 
- Control joint angles
  - make and send a topic to `/robotis/set_joint_states`

Regards,
Kayman

ROBOTIS-Kayman, 2019-03-19:
I found a bug. `op3_direct_control` module make a own trajectory using a thread in the callback function.
when a new topic is subscribed before finishing a thread, it happened. I fixed it, please update a source.

I'm sorry for your inconvenience.
Thanks,
Kayman

yuan0623, 2019-03-19:
> I found a bug. `op3_direct_control` module make a own trajectory using a thread in the callback function.
> when a new topic is subscribed before finishing a thread, it happened. I fixed it, please update a source.
> 
> I'm sorry for your inconvenience.
> Thanks,
> Kayman

Thanks so much for your reply. I've tried `robotis_controller`, and it works great. It respond fast and other joint will not move unexpectedly. I think I will stick on it for this moment!

ROBOTIS-Kayman, 2019-03-20:
It’s my pleasure.
Please open an issue if you have any other questions.

closed 2019-03-20 by ROBOTIS-Kayman

xxchenchen, 2019-08-06:
> Hi, Our lab just purchased a Darwin-OP3. However, I have some issues with direct control mode. Here is the video [link:](https://photos.app.goo.gl/ErftgxeJbRwY678D7). From this video, I have two [...]

Hi Yuan,

Which simulation envrionmet do you use? Gazebo?

xxchenchen, 2019-08-06:
Hi Yuan,

Which simulation envrionmet do you use? Gazebo? Did you solve walking problem?

yuan0623, 2019-08-06:
> Hi Yuan,
> 
> Which simulation envrionmet do you use? Gazebo? Did you solve walking problem? We have walking problem in Gazebo also.

Hi,  I didn't use the simulation software for the issue I've mentioned in this thread. I have never used Gazebo. If you want to use some easy simulation software, I recommend Webots, they have build-in Darwin OP3 model, and it's opensource free software.

xxchenchen, 2019-08-31:
Many thanks