ROBOTIS-GIT/ROBOTIS-OP3 #63: Power supply Intel Nuc and OpenCR

philison, 2019-03-25:
Hi, 
can someone tell me if the Intel Nuc is powered by the 12V@1A output of the OpenCR Board ? 
Would that also work with an Intel Realsense D400 Series camera connected to the Nuc? (or is 12V@1A not enough power?)

Thanks in advance,
Phil

[image]

ROBOTIS-Kayman, 2019-03-26:
Hi,

OP3 uses a `Power Out` connector. Depending on the power supply, it can be over 1A. 
I can not tell you whether it works fine, because we don't have a realsense D400.

[image]

Regards,
Kayman

philison, 2019-03-26:
thank you for the quick response! 
Could you recommend me a battery/power supply to get the most out of the 12V output od the OpenCR ?
Or are there any known limits that I have to obey ? 

Regards,
Phil

ROBOTIS-Kayman, 2019-03-27:
Currently, OP3 uses 12V 5A for SMPS and 11.1V 1800mA (5C) for battery. It is recommended to use at this level, and since the powerout limit is 10A fuse, the fuse may be blown if more current is applied.

Regards, 
Kayman

ROBOTIS-Kayman, 2019-04-12:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-04-12 by ROBOTIS-Kayman

AlfretzNehemia, 2022-01-29:
what kind of cable i should use to connect power line from opencr to nuc?

ROBOTIS-Will, 2022-02-03:
@AlfretzNehemia 
It is recommended to use a cable that can supply 3A at least.
Thanks.

AlfretzNehemia, 2022-02-05:
Thanks for the reply. now, I got it. but I still have a trouble, I'm right
now building a humanoid robot using dynamixel mx-64(upgraded to protocol
2.0). I have tried connect my dynamixel to openCR from TTL port and
uploaded opencr_op3 to openCR. then connect power supply and switch openCR
to on, but why all dynamixel led not blink. When i roslaunch op3_manager
with wiring nuc -> u2d2-> dynamixel -> opencr -> nuc with some modification
in (.robot) file, error occur like this Torque on DXLs! [[RxPacketError]
Unknown error code!]. May you help me solve this issue?

ROBOTIS-Will, 2022-02-09:
Hi @AlfretzNehemia 

The OpenCR should be properly connected to the NUC via U2D2.
The FET on the OpenCR will connect / disconnect the power to DYNAMIXEL, and this FET is controlled by the op3_manager node.
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/src/op3_manager.cpp#L183
If the FET is successfully opened, DYNAMIXEL will flicker once.
Thanks.

AlfretzNehemia, 2022-02-13:
So, there are two cable lines that connected to U2D2. First connect to
dynamixel and second to openCR. Is it right?

ROBOTIS-Will, 2022-02-15:
As long as the U2D2, OpenCR and DYNAMIXEL are connected in the same TTL network, they should be fine.
Since there's only one TTL port on the U2D2, you will need additional board like [this](http://en.robotis.com/shop_en/item.php?it_id=903-0142-000)