ROBOTIS-GIT/ROBOTIS-OP3 #64: Python error when launching manager

haritsahm, 2019-03-27:
I got this error when trying to launch the manager. Is this caused by the cpp program or the ros python?

```
roslaunch op3_manager alfarobi_manager.launch 
... logging to /home/alfarobi/.ros/log/4c92a3d0-50b2-11e9-85e7-3576f2f1f2ec/roslaunch-alfarobi-20180.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 681, in _recurse_load
    self._rosparam_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 239, in _rosparam_tag
    self.load_rosparam(context, ros_config, cmd, param, file, value, verbose=verbose, subst_function=subst_function)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py", line 430, in load_rosparam
    self.add_param(ros_config, full_param, data, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py", line 357, in add_param
    self.add_param(ros_config, ns_join(param_name, k), v, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py", line 357, in add_param
    self.add_param(ros_config, ns_join(param_name, k), v, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/names.py", line 172, in ns_join
    if is_private(name) or is_global(name):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosgraph/names.py", line 133, in is_private
    return name and name[0] == PRIV_NAME
TypeError: 'int' object has no attribute '__getitem__'

```

ROBOTIS-Kayman, 2019-03-28:
Usually this happens when user create the incorrect launch file. 
(ex. parameters are incorrectly written or the grammar is wrong)

Could you check the details in your launch file?

haritsahm, 2019-03-28:
It's my fault, I incorrectly set the parameters in yaml

closed 2019-03-28 by haritsahm