ROBOTIS-GIT/ROBOTIS-OP3 #65: error when try op3_manager with

GramJr, 2019-03-27:
hi
i try op3 manager with standart op3, but i've checked for updates (from R+Manager on Windows) all servos and updated all of it to version 42 (from 40). 

And now op3 manager starts with this errors

process[op3_manager-1]: started with pid [3424]
process[op3_localization-2]: started with pid [3425]
[ INFO] [1553715380.459634064]: manager->init
[ERROR] [1553715380.568218833]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1553715380.602368186]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1553715380.636528690]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1553715380.670664274]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1553715380.704805024]: Torque on DXLs! [[RxPacketError] Unknown error code!]
[ERROR] [1553715380.839121352]: Fail to control LED [[RxPacketError] Unknown error code!]
/dev/ttyUSB0 added. (baudrate: 2000000)
(/dev/ttyUSB0) [ID:  1]         XM-430 added.
(/dev/ttyUSB0) [ID:  2]         XM-430 added.
(/dev/ttyUSB0) [ID:  3]         XM-430 added.
(/dev/ttyUSB0) [ID:  4]         XM-430 added.
(/dev/ttyUSB0) [ID:  5]         XM-430 added.
(/dev/ttyUSB0) [ID:  6]         XM-430 added.
(/dev/ttyUSB0) [ID:  7]         XM-430 added.
(/dev/ttyUSB0) [ID:  8]         XM-430 added.
(/dev/ttyUSB0) [ID:  9]         XM-430 added.
(/dev/ttyUSB0) [ID: 10]         XM-430 added.
(/dev/ttyUSB0) [ID: 11]         XM-430 added.
(/dev/ttyUSB0) [ID: 12]         XM-430 added.
(/dev/ttyUSB0) [ID: 13]         XM-430 added.
(/dev/ttyUSB0) [ID: 14]         XM-430 added.
(/dev/ttyUSB0) [ID: 15]         XM-430 added.
(/dev/ttyUSB0) [ID: 16]         XM-430 added.
(/dev/ttyUSB0) [ID: 17]         XM-430 added.
(/dev/ttyUSB0) [ID: 18]         XM-430 added.
(/dev/ttyUSB0) [ID: 19]         XM-430 added.
(/dev/ttyUSB0) [ID: 20]         XM-430 added.
(/dev/ttyUSB0) [ID:200]        OPEN-CR added.
[ERROR] [1553715381.271926602]: Fail to load offset yaml.
[ WARN] [1553715381.580106068]: Head control - angle unit : 30
[ERROR] [1553715382.158283887]: [RobotisController] first bulk read fail!!
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info$  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[op3_manager-1] process has died [pid 3424, exit code -6, cmd /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager$log file: /home/robotis/.ros/log/60334aba-50c7-11e9-9756-a0c5899b41a0/op3_manager-1*.log

I want to thank you for your assistance with these cases so far.

GramJr, 2019-03-27:
Is it normal that model number was changed from 1000 to 1020 for all servos when i update firmware?

ROBOTIS-Kayman, 2019-03-28:
Hi,

I think the opencr firmware has failed in updating the dxl firmware. Could you refer to the [e-manual](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#op3) to recover the opencr firmware?

The rest of the questions will be answered after confirmation.

Regards,
Kayman

ROBOTIS-Kayman, 2019-03-28:
Dynamixels of OP3 are XM430-W350. 
Model number 1020 is the correct value.
The model number can not be changed through firmware update. 
Only recovery is possible.

Thanks,
Kayman

GramJr, 2019-03-28:
I have already updated firmware on opencr more than 3 times, but from windows. I'll try to do this from linux. (already read this issue - https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/issues/50). I've even rebuilt all op3, replaced opencr (to new one), restored nuc from image and then reinstalled and catkin_make all catkin_ws.

GramJr, 2019-03-28:
Debug information from opencr
0	 Model_Number    	 29696	 0x7400
2	 Firmware_Version	 2	 0x2
3	 ID              	 200	 0xC8
4	 Baud            	 4	 0x4
5	 Return_Delay_Time 	 0	 0x0
16	 Status_Return_Level 	 2	 0x2
18	 Roll_Offset     	 0	 0x0
20	 Pitch_Offset    	 0	 0x0
22	 Yaw_Offset      	 0	 0x0
24	 Dynamixel_Power 	 1	 0x1
25	 LED             	 0	 0x0
26	 LED_RGB         	 992	 0x3E0
28	 Buzzer          	 0	 0x0
30	 Button          	 0	 0x0
31	 Voltage         	 124	 0x7C
32	 Gyro_X          	 16	 0x10
34	 Gyro_Y          	 40	 0x28
36	 Gyro_Z          	 14	 0xE
38	 Acc_X           	 6625	 0x19E1
40	 Acc_Y           	 1542	 0x606
42	 Acc_Z           	 15318	 0x3BD6
44	 Roll            	 59	 0x3B
46	 Pitch           	 -234	 0xFFFFFF16
48	 Yaw             	 -286	 0xFFFFFEE2
50	 IMU_Control     	 0	 0x0

ROBOTIS-Kayman, 2019-03-29:
Hi,

Did it work after rebuilt?
OPENCR looks good now.
Model number and firmware version, id, baud of OPENCR are fine.
If not, could you show me the log of op3_manager?

Regards,
Kayman

GramJr, 2019-03-29:
Hi, sure, here it is

GramJr, 2019-03-29:
Hi, here it is
[master.log](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/3021277/master.log)
[roslaunch-robotis-op3-4169.log](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/3021278/roslaunch-robotis-op3-4169.log)
[rosout.log](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/3021279/rosout.log)
[rosout-1-stdout.log](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/3021280/rosout-1-stdout.log)

ROBOTIS-Kayman, 2019-03-29:
Hi, 

I looked at your log and it still seems that OPENCR is not responding.
Would you like to test it with dxl_monitor?
There is a program called dxl_monitor in the example of Dynamixel. 
After connecting to OP3 with ssh, please follow the instructions below to see if OPENCR responds properly.
- Installation method
  - build library
  $ cd ~ / catkin_ws / DynamixelSDK / c ++ / build / linux64
  $ make
  $ sudo make install

  - build dxl_monitor and run
  $ cd ~ / catkin_ws / DynamixelSDK / c ++ / example / dxl_monitor / linux64
  $ make
  $ ./dxl_monitor

  - test
    - Change baud
    [CMD] baud 2000000
    - ping test
    [CMD] ping 200

[image]

GramJr, 2019-03-29:
Done, ping to opencr (id 200) failed

GramJr, 2019-03-29:
I have tested with dxl_monitor from opencr dxl bus, and it works well!   Opencr without nuc succeed to ping, to write and read to XM430.
[image]
[image]

GramJr, 2019-03-29:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

GramJr, 2019-03-29:
1.4.X core for opencr was the reason of this issue. Now i use 1.3.2 core and all works fine. 
If it possible, i want to know why this happens. :)

OpusK, 2019-04-01:
Hi, @GramJr 

I will check about this.
And if I know the exact cause, I'll let you know.

OpusK, 2019-04-01:
@GramJr 

I've found a problem related to UART DMA function.
It was added in 1.4.0, but I found that it was not applied to the OP3 example.

I will update this with other features and will release it within this week.
Sorry for the inconvenience. And thanks for giving me more debugging info :)

GramJr, 2019-04-01:
@OpusK 
Ok, thanks. Can I ask some questions about OP3 by mail, which is listed in the profile?

OpusK, 2019-04-02:
@GramJr 

If possible, it would be nice for us and other users to leave the issue on GitHub.
But, if your question is difficult to address here, please ask me about OpenCR's Firmware and ask @ROBOTIS-Kayman for OP3's software.

GramJr, 2019-04-02:
@OpusK
Thank you!

closed 2019-04-02 by GramJr

OpusK, 2019-04-02:
@GramJr 

I just released version [1.4.6 of OpenCR](https://github.com/ROBOTIS-GIT/OpenCR/releases/tag/1.4.6)!

AlfretzNehemia, 2021-08-22:
hello guys, i have same problem here. i use u2d2->opencr -> dynamixel mx-64. when tried to ping opencr, it always fails. when i roslaunch op3_manager, always error. i have used op3 code in arduino ide. i hope you can response my problem. Thank you

ROBOTIS-Will, 2021-08-23:
Hi @AlfretzNehemia 
The OP3 is designed specifically for DYNAMIXEL X series and does not compatible with MX series.
Thank you.

AlfretzNehemia, 2021-08-23:
thank you for answering my question. So, do you have any suggestion for me
to run robot with dynamixel mx,ud2d, and subcontroller opencr? and i still
have one
question, why my opencr can't success ping when i test dxl_monitor?

ROBOTIS-Will, 2021-08-30:
@AlfretzNehemia 
You may try upgrading to [MX-64(2.0)](https://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/) firmware via DYNAMIXEL Wizard 2.0.
MX series with Protocol 2.0 firmware is very similar to X series.
Also you'll need to make some modifications in the source such as [OP3.robot](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/OP3.robot) file where specific DYNAMIXEL is defined.

In regard with the ping issue, please try searching the opencr issue page, for example, https://github.com/ROBOTIS-GIT/OpenCR/issues/147

Thanks!