ROBOTIS-GIT/ROBOTIS-OP3 #66: Program stopped running after disconnecting LAN connection to laptop

BagaskaraPutra, 2019-04-06:
Hi,
I ran the ROBOTIS_OP3 soccer.cpp program on the robot while connecting it to the laptop via SSH LAN cable. But when the LAN cable is disconnected, the program stopped running. Is there any way to make the program still run even though the LAN cable is disconnected from the robot?
Thank you

ROBOTIS-Kayman, 2019-04-08:
Hi~

If you run the program via ssh, execution terminates when the network connection is lost.
Please try it as background run. please refer to the following [link](https://www.maketecheasier.com/run-bash-commands-background-linux/).

Regards,
Kayman

BagaskaraPutra, 2019-04-11:
> Hi~
> 
> If you run the program via ssh, execution terminates when the network connection is lost.
> Please try it as background run. please refer to the following [link](https://www.maketecheasier.com/run-bash-commands-background-linux/).
> 
> Regards,
> Kayman

Thanks for the link. But, after trying all the commands from the link, the problem still exists. The program still terminates when LAN connection is lost. Do you have any other suggestions?
Thank you

ROBOTIS-Kayman, 2019-04-12:
Sorry,
I don't know any other solution.

Regards,
Kayman

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman