ROBOTIS-GIT/ROBOTIS-OP3 #67: question about auto-fall recovery

yuan0623, 2019-04-10:
Hi, I have a simple question, does OP3 has the ability to recover itself from falling on ground, in other word, once the OP3 falls on the ground, can it automatically stand up? I didn't find anything in the manual about that.

ROBOTIS-Kayman, 2019-04-10:
Hi,

OPENCR on OP3 has a gyro and acceleration sensor. 
We use this value to check that the robot has fallen in the OP3 soccer demo. 
If it falls, it will take action for getting up. 
Please refer to source code.
 - [check for falls](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Demo/blob/77d76182b39c8752388bda48d484cad5f9fedf70/op3_demo/src/soccer/soccer_demo.cpp#L438)
 - [action to get up](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Demo/blob/77d76182b39c8752388bda48d484cad5f9fedf70/op3_demo/src/soccer/soccer_demo.cpp#L585)

Regards, 
Kayman

yuan0623, 2019-04-10:
Thanks for your fast reply!
I see, we only need to publish a integer to `/robotis/action/page_num`, but is there a motion index list? I didn't see any in the source code. I'm not familiar with c++, I'm going to use Python to publish it. 
I really have no idea about how much is the `is_grass_ ? GetUpFront + ForGrass : GetUpFront` and `is_grass_ ? GetUpBack + ForGrass : GetUpBack`

yuan0623, 2019-04-11:
thank you so much! I think I figured it out!

closed 2019-04-11 by yuan0623