ROBOTIS-GIT/ROBOTIS-OP3 #68: Walking stability for op3_walking_module

VasilyevGeorgy, 2019-04-10:
Hi!
I develop custom walking module for ROBOTIS OP3. 
In order to check OP3's calibration I've run op3_walking_module.
So I've tried it in on linoleum floor and on the table top.
In both cases the robot falls (it's landing its feet not planar for ground surface).

There is also op3_tuning_module in package with several tune_poses.

So, I got some questions:
1. May I use op3_tuning_module somehow in order to calibrate OP3's walking?
2. Is there the way how to set up op3_walking_module achieve a stable walk with it?
3.  How can I calibrate OP3?

Regards,
George

ROBOTIS-Kayman, 2019-04-12:
Hi,

op3_tuning_module is for calibration of neutrol situation.
op3_tuning_module is for hardware calibration.
We provide the program for the tuning of op3's walking.
Please refer to following links.
- [gui for walking tuner](http://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-use-walking-tuner).
- [op3_walking_module](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#op3_walking_module)

Regards,
Kayman

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman