ROBOTIS-GIT/ROBOTIS-OP3 #69: question about optimal parameter for turning left and right, side-walking in op3_walking_module

yuan0623, 2019-04-13:
Hi
My project require the robot turn at the fixed position (or almost fixed position) as well as side-walking . I try to use `op3_walking_module` to set the yaw angle to do it. However, it behaves really bad. The foot step on each other and the robot is not able to stabilize itself.  I see the video that OP3 is totally capable of turning and side walking, May I ask is there any place that I can get the parameters to do the turning and side-walking? 
Thanks in advance!!
Best
Yuan

ROBOTIS-Kayman, 2019-04-15:
Hi,

Please refer to following links. 
you can find the appropriate parameters in trial and error.
- How to tune walking parameters : [gui for walking tuner](http://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-use-walking-tuner).
- walking parameters : [op3_walking_module](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#op3_walking_module)

Regards,
Kayman

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman