ROBOTIS-GIT/ROBOTIS-OP3 #70: Using foot sensors with op3

GramJr, 2019-04-15:
Hello again,
1. I don't know where i can write about CM-530, but i have found small bug (OUT Ports 1-6 didn't  works with HIGH value when i set HEX value ...0011, or both LEDs to ON).

2. I need to connect foot-force sensors to op3, maybe you can tell me more about variables used in balance control (https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/9e306bbfbf769908dab240d09f40208670762bab/op3_balance_control/src/op3_balance_control.cpp)? At what topic i need to publish data to connect my sensors to balance control?

ROBOTIS-Kayman, 2019-04-15:
Hi,

1. I do not understand your question. Are you talking about OPENCR?

2. I'm sorry, but it will be difficult to explain the source code line by line.

3. THORMANG3 uses imu and ft sensor data. The source code of OP3's walking is a reduced version of THORMANG3, so it is recommended to analyze [the walking source code of THORMANG3](http://emanual.robotis.com/docs/en/platform/thormang3/thormang3_ros_packages/#thormang3_walking_module). 

4. In case of THORMANG3, the ft sensor is connected to the external port of the ankle dynamixels.
  `thormang3_manager` read the analog value and [`feet_ft_module`](http://emanual.robotis.com/docs/en/platform/thormang3/thormang3_ros_packages/#thormang3_feet_ft_module) makes it as ft data.

Regards,
Kayman

GramJr, 2019-04-15:
I'm talking about this controller - http://support.robotis.com/en/product/controller/cm530.htm

Ok, i need time to read information about thormang ft sensors.

ROBOTIS-Kayman, 2019-04-15:
I see.
It was not a product to use in OP3, so I made a mistake.
I will contact the person i charge about the case.

Regards, 
Kayman

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman