ROBOTIS-GIT/ROBOTIS-OP3 #72: Memory error while op3_manager is running

VasilyevGeorgy, 2019-08-08:
Hi!
After we fixed OpenCR board we got new error while op3_manager is running.
```
[ INFO] [1565263982.275285646]: [start] send trajectory
[ INFO] [1565263986.282749598]: [end] send trajectory
*** Error in `/home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager': free(): invalid next size (fast): 0x00007ff93c002920 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ff97fb847e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7ff97fb8d37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7ff97fb9153c]
/home/robotis/catkin_ws/devel/lib/libdynamixel_sdk.so(_ZN9dynamixel22Protocol2PacketHandler6readRxEPNS_11PortHandlerEhtPhS3_+0x111)[0x7ff980e5a603]
/home/robotis/catkin_ws/devel/lib/libdynamixel_sdk.so(_ZN9dynamixel13GroupBulkRead8rxPacketEv+0x110)[0x7ff980e610b2]
/home/robotis/catkin_ws/devel/lib/librobotis_controller.so(_ZN17robotis_framework17RobotisController7processEv+0x357)[0x7ff982b3133b]
/home/robotis/catkin_ws/devel/lib/librobotis_controller.so(_ZN17robotis_framework17RobotisController11timerThreadEPv+0x206)[0x7ff982b2f8a2]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7ff97f0546ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7ff97fc1441d]
======= Memory map: ========
00400000-00438000 r-xp 00000000 08:02 5248458                            /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager
00638000-00639000 r--p 00038000 08:02 5248458                            /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager
00639000-0063a000 rw-p 00039000 08:02 5248458                            /home/robotis/catkin_ws/devel/lib/op3_manager/op3_manager
023d9000-0272f000 rw-p 00000000 00:00 0                                  [heap]
7ff930000000-7ff930021000 rw-p 00000000 00:00 0 
7ff930021000-7ff934000000 ---p 00000000 00:00 0 
7ff9377ff000-7ff937800000 ---p 00000000 00:00 0 
7ff937800000-7ff938000000 rw-p 00000000 00:00 0 
7ff938000000-7ff938021000 rw-p 00000000 00:00 0 
7ff938021000-7ff93c000000 ---p 00000000 00:00 0 
7ff93c000000-7ff93c021000 rw-p 00000000 00:00 0 
7ff93c021000-7ff940000000 ---p 00000000 00:00 0 
7ff940000000-7ff940021000 rw-p 00000000 00:00 0 
7ff940021000-7ff944000000 ---p 00000000 00:00 0 
7ff944000000-7ff944021000 rw-p 00000000 00:00 0 
7ff944021000-7ff948000000 ---p 00000000 00:00 0 
7ff948000000-7ff948021000 rw-p 00000000 00:00 0 
7ff948021000-7ff94c000000 ---p 00000000 00:00 0 
7ff94c000000-7ff94c021000 rw-p 00000000 00:00 0 
7ff94c021000-7ff950000000 ---p 00000000 00:00 0 
7ff950000000-7ff950021000 rw-p 00000000 00:00 0 
7ff950021000-7ff954000000 ---p 00000000 00:00 0 
7ff954000000-7ff95402a000 rw-p 00000000 00:00 0 
7ff95402a000-7ff958000000 ---p 00000000 00:00 0 
7ff958000000-7ff958021000 rw-p 00000000 00:00 0 
7ff958021000-7ff95c000000 ---p 00000000 00:00 0 
7ff95c000000-7ff95c021000 rw-p 00000000 00:00 0 
7ff95c021000-7ff960000000 ---p 00000000 00:00 0 
7ff960000000-7ff960021000 rw-p 00000000 00:00 0 
7ff960021000-7ff964000000 ---p 00000000 00:00 0 
7ff964000000-7ff964021000 rw-p 00000000 00:00 0 
7ff964021000-7ff968000000 ---p 00000000 00:00 0 
7ff968000000-7ff968021000 rw-p 00000000 00:00 0 
7ff968021000-7ff96c000000 ---p 00000000 00:00 0 
7ff96c5b0000-7ff96c5b1000 ---p 00000000 00:00 0 
7ff96c5b1000-7ff96cdb1000 rw-p 00000000 00:00 0 
7ff96cdb1000-7ff96cdb2000 ---p 00000000 00:00 0 
7ff96cdb2000-7ff96d5b2000 rw-p 00000000 00:00 0 
7ff96d5b2000-7ff96d5b3000 ---p 00000000 00:00 0 
7ff96d5b3000-7ff96ddb3000 rw-p 00000000 00:00 0 
7ff96ddb3000-7ff96ddb4000 ---p 00000000 00:00 0 
7ff96ddb4000-7ff96e5b4000 rw-p 00000000 00:00 0 
7ff96e5b4000-7ff96e5b5000 ---p 00000000 00:00 0 
7ff96e5b5000-7ff96edb5000 rw-p 00000000 00:00 0 
7ff96edb5000-7ff96edb6000 ---p 00000000 00:00 0 
7ff96edb6000-7ff96f5b6000 rw-p 00000000 00:00 0 
7ff96f7ff000-7ff96f800000 ---p 00000000 00:00 0 
7ff96f800000-7ff970000000 rw-p 00000000 00:00 0 
7ff970000000-7ff970021000 rw-p 00000000 00:00 0 
7ff970021000-7ff974000000 ---p 00000000 00:00 0 
7ff974571000-7ff974572000 ---p 00000000 00:00 0 
7ff974572000-7ff974d72000 rw-p 00000000 00:00 0 
7ff974d72000-7ff974d73000 ---p 00000000 00:00 0 
[... 238 lines cut ...]
7ff98278d000-7ff98278f000 r--p 000a7000 08:02 4851468                    /home/robotis/catkin_ws/devel/lib/libop3_base_module.so
7ff98278f000-7ff982793000 rw-p 000a9000 08:02 4851468                    /home/robotis/catkin_ws/devel/lib/libop3_base_module.so
7ff982793000-7ff982808000 r-xp 00000000 08:02 4851472                    /home/robotis/catkin_ws/devel/lib/libopen_cr_module.so
7ff982808000-7ff982a08000 ---p 00075000 08:02 4851472                    /home/robotis/catkin_ws/devel/lib/libopen_cr_module.so
7ff982a08000-7ff982a09000 r--p 00075000 08:02 4851472                    /home/robotis/catkin_ws/devel/lib/libopen_cr_module.so
7ff982a09000-7ff982a0c000 rw-p 00076000 08:02 4851472                    /home/robotis/catkin_ws/devel/lib/libopen_cr_module.so
7ff982a0c000-7ff982bc4000 r-xp 00000000 08:02 4851478                    /home/robotis/catkin_ws/devel/lib/librobotis_controller.so
7ff982bc4000-7ff982dc3000 ---p 001b8000 08:02 4851478                    /home/robotis/catkin_ws/devel/lib/librobotis_controller.so
7ff982dc3000-7ff982dc7000 r--p 001b7000 08:02 4851478                    /home/robotis/catkin_ws/devel/lib/librobotis_controller.so
7ff982dc7000-7ff982dd0000 rw-p 001bb000 08:02 4851478                    /home/robotis/catkin_ws/devel/lib/librobotis_controller.so
7ff982dd0000-7ff982df6000 r-xp 00000000 08:02 3145962                    /lib/x86_64-linux-gnu/ld-2.23.so
7ff982f81000-7ff982fa6000 rw-p 00000000 00:00 0 
7ff982fdd000-7ff982ff5000 rw-p 00000000 00:00 0 
7ff982ff5000-7ff982ff6000 r--p 00025000 08:02 3145962                    /lib/x86_64-linux-gnu/ld-2.23.so
7ff982ff6000-7ff982ff7000 rw-p 00026000 08:02 3145962                    /lib/x86_64-linux-gnu/ld-2.23.so
7ff982ff7000-7ff982ff8000 rw-p 00000000 00:00 0 
7ffd965dc000-7ffd965fd000 rw-p 00000000 00:00 0                          [stack]
7ffd967df000-7ffd967e1000 r--p 00000000 00:00 0                          [vvar]
7ffd967e1000-7ffd967e3000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
```
What could cause this error?

Regards,
George

ROBOTIS-Kayman, 2019-08-09:
Hi,

From the backtrace, I think there is a problem when bulk reading from the process () function of the robotis_controller.
If you modify the product arbitrarily, it is difficult to help debug the problem that may occur.

Regards,
Kayman

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will