ROBOTIS-GIT/ROBOTIS-OP3 #73: op3 camera

xxchenchen, 2019-09-02:
Hi
How can we run the camera module in gazebo? Is it possible to generate a environment using the op3 camera in gazebo?

ROBOTIS-Kayman, 2019-09-05:
Hi,
you can use the camera in gazebo.
1. gazebo
```
$ roslaunch op3_gazebo robotis_world.launch
```

2. op3_manager for gazebo
```
$ roslaunch op3_manager op3_gazebo.launch
```

3. display in rviz
```
$ roslaunch op3_description op3_display.launch
```

[image]
[image]

xxchenchen, 2019-09-20:
Thanks for your answer.

closed 2019-09-20 by xxchenchen

ahmaddidiks, 2020-09-22:
I did roslaunch op3_description op3_display.launch, but got some orrors. I've installed joint-state-publisher-gui but I got this,
[WARN] [1600796056.614672]: The 'use_gui' parameter was specified, which is deprecated.  We'll attempt to find and run the GUI, but if this fails you should install the 'joint_state_publisher_gui' package instead and run that.  This backwards compatibility option will be removed in Noetic.