ROBOTIS-GIT/ROBOTIS-OP3 #74: op3 arm joint torque.

xxchenchen, 2019-09-02:
Hi,

How can we obtain torque values relating to the arm joints of robotis-op3?

could you please write me 2-3 lines of code?

Thanks

ROBOTIS-Kayman, 2019-09-05:
Hi, 

There is no torque in x series. you can read the current only. 
If you want to obtain the present current. I suggest 2 ways to get it.

1. using robotis_framework : refer to [thormang_p_manager](https://github.com/ROBOTIS-GIT/ROBOTIS-THORMANG-P-MPC/tree/master/thormang3_p_manager) and [thormang3_alarm_module](https://github.com/ROBOTIS-GIT/ROBOTIS-THORMANG-P-MPC/tree/master/thormang3_alarm_module)
  - change `robot_model` in manager
  - read the present current in process function of motion_module or sensor_module.

2. using dynamixel_sdk : refer to [sdk e-manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_read_write_protocol_2_0/#cpp-read-write-protocol-20)
```cpp
#include "dynamixel_sdk.h"
using namespace dynamixel;

const int BAUD_RATE = 2000000;
const double PROTOCOL_VERSION = 2.0;
const int DEFAULT_DXL_ID = 1;
const int CURRENT_CTRL_TABLE = 126;

PortHandler *port_handler = (PortHandler *) PortHandler::getPortHandler(""/dev/ttyUSB0");
port_handler->setBaudRate(BAUD_RATE);
PacketHandler *packet_handler = PacketHandler::getPacketHandler(PROTOCOL_VERSION);
uint16_t current = 0;
// port handler, dynamixel id, control table id
packet_handler->read2ByteTxRx(port_handler, DEFAULT_DXL_ID , CURRENT_CTRL_TABLE , &current);

```

Regards,
Kayman

xxchenchen, 2019-09-20:
Thanks for your answer.

xxchenchen, 2019-09-23:
Hİ,

we did it. However, when we reads the current values, manager cancels or we cant control by keyboard the op3.

Why? Could you please say it? thanks

ROBOTIS-Kayman, 2019-09-26:
Hi,

Can you tell me more details?
Which of the above two methods did you use?
Attach a terminal log image for the problem situation.

Regards,
Kayman

xxchenchen, 2019-10-04:
Hi,
We use the second method. We control the robot by keyboard. The robot grasp an object. We want to obtain force/torque/current knowledge.
[image]
[image]

xxchenchen, 2019-10-11:
Are you there?

ROBOTIS-Kayman, 2019-10-14:
I'm sorry I'm late.

If you are using robotis_framework in op3_manager to control OP3, you can't use the second method.
Only one program can control OP3 at a time. If you use op3_manager, you should use the first method.

Edit `op3_manager/config/OP3.robot` and add` present_current` to `BULK READ ITEMS`.

In the `process` function of your MotionModule class, read and use the following:
```
       std :: string present_current_name = dxl-> present_current_item _-> item_name_;
       uint16_t present_current_data = dxl-> dxl_state _-> bulk_read_table_ [present_current_name];
       int16_t data = present_current_data;
```

Please refer to Thormang Code([robot file](https://github.com/ROBOTIS-GIT/ROBOTIS-THORMANG-P-MPC/blob/master/thormang3_p_manager/config/THORMANG3_P.robot), [alarm_module](https://github.com/ROBOTIS-GIT/ROBOTIS-THORMANG-P-MPC/blob/52e64a08d4a759c2eb93530b25d9a1bd10a58442/thormang3_alarm_module/src/alarm_module.cpp#L145)).

Regards, 
Kayman

xxchenchen, 2019-10-15:
Many thanks for reply. Just now we will try it.

xxchenchen, 2019-10-18:
Hi,
We receive error. We need to the other lines also: 
      
[image]
std :: string present_current_name = dxl-> present_current_item _-> item_name_;
       uint16_t present_current_data = dxl-> dxl_state _-> bulk_read_table_ [present_current_name];
       int16_t data = present_current_data;

Could you please say them
? Many thanks

xxchenchen, 2019-10-18:
Please answer it.

ROBOTIS-Kayman, 2019-10-21:
Please refer to [Alarm_module](https://github.com/ROBOTIS-GIT/ROBOTIS-THORMANG-P-MPC/blob/52e64a08d4a759c2eb93530b25d9a1bd10a58442/thormang3_alarm_module/src/alarm_module.cpp#L145) of THORMANG3.
Code from line 145 to line 165 of `process` function is related to getting present current.

xxchenchen, 2019-10-21:
We couldn't do it. Unable to cope with different mistakes. Thanks

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will