ROBOTIS-GIT/ROBOTIS-OP3 #77: Now we want to do both walk and transport a object.

xxchenchen, 2019-10-21:
Hi Kayman,
We can run the robot. We can capture objects. Now we want to do both together. How to walk and carry objects together. Let grasp the object and let us move it and leave it somewhere else. Could you please a little information about this information? Thanks

ROBOTIS-Kayman, 2019-10-22:
Hi,

If you use 'op3_manager', you can control the upper and lower body of robotis-op3 separately.
Please refer to this issue(#40).

And if you use other package to control the upper body of robotis-op3 to capture the objects, you can merge the system by creating a motion module for the upper body. refer to the [e-manual](http://emanual.robotis.com/docs/en/software/robotis_framework_packages/tutorials/#creating-new-motion-module).

Regards,
Kayman

xxchenchen, 2019-10-22:
Many thanks Kayman. We will try it.

closed 2020-12-11 by ROBOTIS-Will