ROBOTIS-GIT/ROBOTIS-OP3 #78: Create new motion and play it continuesly

fuad99, 2019-11-06:
How to make new motion? because I tried to read the documentation and it's make me confuse. Can I make new motion with another way?

ROBOTIS-Kayman, 2019-11-11:
Hi,

I'm sorry but there are no other ways to make motions.
I hope this helps.
- type `help` to get the help menu.
```
exit               Exits the program.
re                 Refreshes the screen.
b                  Move to previous page.
n                  Move to next page.
page [index]       Move to page [index].
list               View list of pages.
new                Clears data of current page and initializes page.
copy [index]       Copy data from page [index].
set [value]        Sets value on cursor [value].
save               Saves changes.
play               Motion playback of current page.
playboth [index]   Motion playback of current page with [index] mp3.
g [index]          Motion playback of STP[index].
name               Name for current page or changes the name of current page.
time               Change time base playing.
speed              Change speed base playing.
w [index]          Overwrites data from STP[index] with STP7.
i                  Inserts data from STP7 to STP0. 
                   Moves data from STP[x] to STP[x+1].
i [index]          Inserts data from STP7 to STP[index]. 
                   Moves data from STP[index] to STP[index+1].
m [index] [index2] Moves data from [index] to [index2] step.
d [index]          Deletes data from STP[index]. 
                   Pushes data from STP[index] to STP[index-1].
mlr [index]        Mirror the left side value to the right of STP[index]. 
mrl [index]        Mirror the right side value to the left of STP[index]. 
ms [index]         Switch the values between left and right of STP[index]. 
on/off             Turn On/Off torque from ALL actuators.
on/off [index1] [index2] ...  
                   turns On/Off torque from ID[index1] ID[index2]...
```

- shoutcut
  - `Arrow keys` : You can move the cursor using the arrow keys.
  - `[` : Decrease value by 1
  - `]` : Increase value by 1
  - `{` : Decrease value by 10
  - `}` : Increase value by 10
  - `Space Bar` : Torque on/Off of the joint pointed by the cursor. It only works when there is a cursor in Step7(STP7).
- The op3 action editor is a program similar to R + motion. Please refer to [this manual](http://emanual.robotis.com/docs/en/software/rplus1/motion/#pose-editing).
  - Angle: 4096 resolution (2048 is 0 degrees)
  - Pose of Robot: STP7
  
Regards,
Kayman

fuad99, 2019-11-11:
okay, if i'm done create motion in many page, can i play it using start button(like demo)?

ROBOTIS-Kayman, 2019-11-12:
yes, you can make a program like demo.
If you run op3_manager, you can get the button status by the message.
(start, start_long, mode, mode_lone, user, user_long)
This is [api for button](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#ros-api-7)

Then you can enable the action_module and send the action page number to play the motion.
This is [api of action_module](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#ros-api)

Regards,
Kayman

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will