ROBOTIS-GIT/ROBOTIS-OP3 #81: Add servo

waktep, 2019-11-14:
hello 
[URGENT CASE]
i would like to add more dynamixel (XM430) on robotis op3, is  there any procedure or step that should be proceed ? please mention me the option and description toward that problem.

ROBOTIS-Kayman, 2019-11-15:
Hi,

If you use op3_manager, you have to do like the below.
- Set servo 
  - baudrate : 2M
  - ID : set the ID that don't overlap with the existing servo.
- modify the [ROBOT file](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/OP3.robot)
    - add servo to the ROBOT file with new joint name and ID.
- modify the [dxl_init_OP3.yaml](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/dxl_init_OP3.yaml) for initial setting to servo.
- modify the motion_module you want to use.
  (ex. op3_action_module, op3_base_module, op3_walking_module, etc)
- [other tool program](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3-Tools) you want to use.
  (ex. op3_gui_demo, op3_action_editor, etc)

Regards,
Kayman

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will

raypsaw, 2021-06-17:
If I want to use op3_base_module, is only the .yaml file edited or is there something else?

ROBOTIS-Will, 2021-06-18:
Depending on the number of additional modules, you might need to additionally modify the source code such as MAX_JOINT_ID
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_kinematics_dynamics/include/op3_kinematics_dynamics/op3_kinematics_dynamics_define.h#L22