ROBOTIS-GIT/ROBOTIS-OP3 #83: can I directly control the torque of the joints?

yuan0623, 2019-11-29:
Hi, may I know if there is a way to directly control the torque of the joints? In our dynamic model, the input is torque rather than position. Thus, we need to command torque to the joint. However, I didn't find such a way to do it. Thanks so much for your response!

ROBOTIS-Kayman, 2019-11-29:
Hi,

Sorry, there is no way to control the torque of the joints.
If you want, you have to develop the software using Dynamixel SDK.
  - [Github](https://github.com/ROBOTIS-GIT/DynamixelSDK)
  - [E-manual](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_read_write_protocol_2_0/#cpp-protocol-20)

Regards,
Kayman

fuad99, 2019-11-29:
Maybe you can try using Dynamixel Workbench application to control each Dynamixel

yuan0623, 2019-12-01:
> Hi,
>
> Sorry, there is no way to control the torque of the joints.
> If you want, you have to develop the software using Dynamixel SDK.
>
> * [Github](https://github.com/ROBOTIS-GIT/DynamixelSDK)
> * [...]

thank you sooooo much!

yuan0623, 2019-12-01:
> Maybe you can try using Dynamixel Workbench application to control each Dynamixel

thank you, bro!

closed 2019-12-01 by yuan0623