ROBOTIS-GIT/ROBOTIS-OP3 #84: what's the best way to implement customized gait on Darwin?

yuan0623, 2019-12-05:
Hi. I have a question about how to implement the customized gait on Darwin? 
Current, I publish the joint angle to `/robotis/set_joint_states`. It works, however, the behavior of the robot is kind of jerky, not as smooth as the default one. 
Below are the videos that show the differences between the default gait and customized gait.
This is the [customized one](https://photos.app.goo.gl/EuEJDkNHE69sU7X98);
and this is the [default one](https://photos.app.goo.gl/EgLncmiEuCp7PSqk6)
If you hear the sound of the motor, the customized one is really jerky.
Is there a better way to load the customized trajectory into Darwin to achieve better performance?
Looking forward to your response!
By the way, I'm using python to publish the trajectory, should I use c++ to have a better performance?

Best
Yuan

ROBOTIS-Kayman, 2019-12-06:
Hi,

I think you should create a motion_module using c++.
It is best to run operations and controls in one program.
Controlling robots using topics is difficult to achieve the desired results due to lack of real time.

- [motion_module tutorial](http://emanual.robotis.com/docs/en/software/robotis_framework_packages/tutorials/#creating-new-motion-module)
- [robot_manager tutorial](http://emanual.robotis.com/docs/en/software/robotis_framework_packages/tutorials/#creating-new-robot-manager)
- [op3_manager](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/b2d35a1f545e3db6bb196f4b649dcbc6303e0efb/op3_manager/src/op3_manager.cpp#L241)

Regards,
Kayman

yuan0623, 2019-12-06:
my requirement is at each timestep, one computation is required to obtain the desired position (and velocity) of each joint. Can motion_module help me to achieve that?

ROBOTIS-Kayman, 2019-12-06:
Yes, it can.
For example, looking at the `op3_walking_module`, it calculates the target position of each joint at every control cycle and sends the result to the dynamixel using `robotis_framework(op3_manager)`.
- [op3_walking_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/b2d35a1f545e3db6bb196f4b649dcbc6303e0efb/op3_walking_module/src/op3_walking_module.cpp#L424)

Regards,
Kayman

yuan0623, 2019-12-06:
> Yes, it can.
> For example, looking at the `op3_walking_module`, it calculates the target position of each joint at every control cycle and sends the result to the dynamixel using [...]

Thanks for your response. If I understand it correctly, the `op3_walking_module` is one of the 'motion_module', and I need to create another 'motion_module' to achieve the performance like the one in `op3_walking_module`, am I right?

ROBOTIS-Kayman, 2019-12-06:
That's right, you understood it well.

Regards,
Kayman

closed 2019-12-12 by yuan0623