ROBOTIS-GIT/ROBOTIS-OP3 #85: the connection between body and arm breaks, so I have following questions.

yuan0623, 2019-12-12:
Hi, the darwin accidentally falls onto the ground and breaks the connection screw between the arm and body. The pictures can be seen [here](https://www.dropbox.com/s/s5cx7b35ckpssmr/IMG_20191211_233400.jpg?dl=0) and [here](https://www.dropbox.com/s/4zrzva7e63e9uic/IMG_20191211_233410.jpg?dl=0 ). It seems than the screw breaks inside the motor. I think it's hard to get the broken screw out at this size. So I have following questions:
1.  is it possible for us to buy a new motor and replace the old one? is it easy to do? do we need to do extensive configuration after the replacement?
2.  if we disconnect the motor cable on arm (so that the op3 can walk without arms), can control-manager still function properly? I know at first the control-manager will scan all the hardware first.
Looking forward to your response!
Yuan

ROBOTIS-Kayman, 2019-12-13:
Hi,

I'm sorry to hear about the bad news.
1. I think it's better to send an RMA, but you can change the motor yourself.  follow these instructions.
   1. Prepare a new XM430-W310.
   2. Set the ID and baudrate using [Dynamixal Wizard 2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/).  
      - ID : 1(right should pitch)
      - baudrate : 2M
   3. Open the top and back plates of the chest of the OP3 and replace the motor.  
    (Please note that the wire is connected inside.)

2. If you want to use OP3 with only some motors connected, follow the below
    - modify the [robot file(op3_manager/config/OP3.robot)](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/OP3.robot).
    (e.g. If you want to disconnect the right arm, comment out the right arm in the robot file and change the default joint to another one.)
    ```
    [ control info ]
    control_cycle = 8   # milliseconds

    [ port info ]
    # PORT NAME  | BAUDRATE  | DEFAULT JOINT
    /dev/ttyUSB0 | 2000000   | l_sho_pitch

    [ device info ]
    # TYPE    | PORT NAME    | ID  | MODEL          | PROTOCOL | DEV NAME       | BULK READ ITEMS
    # dynamixel | /dev/ttyUSB0 | 1   | XM430-W350     | 2.0      | r_sho_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 2   | XM430-W350     | 2.0      | l_sho_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    # dynamixel | /dev/ttyUSB0 | 3   | XM430-W350     | 2.0      | r_sho_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 4   | XM430-W350     | 2.0      | l_sho_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    # dynamixel | /dev/ttyUSB0 | 5   | XM430-W350     | 2.0      | r_el           | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 6   | XM430-W350     | 2.0      | l_el           | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 7   | XM430-W350     | 2.0      | r_hip_yaw      | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 8   | XM430-W350     | 2.0      | l_hip_yaw      | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 9   | XM430-W350     | 2.0      | r_hip_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 10  | XM430-W350     | 2.0      | l_hip_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 11  | XM430-W350     | 2.0      | r_hip_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 12  | XM430-W350     | 2.0      | l_hip_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 13  | XM430-W350     | 2.0      | r_knee         | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 14  | XM430-W350     | 2.0      | l_knee         | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 15  | XM430-W350     | 2.0      | r_ank_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 16  | XM430-W350     | 2.0      | l_ank_pitch    | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 17  | XM430-W350     | 2.0      | r_ank_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 18  | XM430-W350     | 2.0      | l_ank_roll     | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 19  | XM430-W350     | 2.0      | head_pan       | present_position, position_p_gain, position_i_gain, position_d_gain
    dynamixel | /dev/ttyUSB0 | 20  | XM430-W350     | 2.0      | head_tilt      | present_position, position_p_gain, position_i_gain, position_d_gain
    sensor    | /dev/ttyUSB0 | 200 | OPEN-CR        | 2.0      | open-cr        | button, present_voltage, gyro_x, gyro_y, gyro_z, acc_x, acc_y, acc_z, roll, pitch, yaw
    ```

Regards,
Kayman

yuan0623, 2019-12-13:
Thank you so much for your information. You have always been so helpful! Wish you had a great weekend!

closed 2019-12-13 by yuan0623