ROBOTIS-GIT/ROBOTIS-OP3 #86: question about lateral balance

yuan0623, 2019-12-17:
Hi, I have a question about the lateral balance while op3 is walking.
I know some people use IMU data to regulate the hip roll joint to prevent the robot from tilting sideway. I'm thinking to try the same thing, however, I have some questions and hope you can help me. In this discussion, *I convert the quaternion to **Euler angles** to fit my convention.*
1. During the walking, the IMU data is quite noisy. I have recorded a video [here](https://drive.google.com/file/d/1zjCxDXROhlwkSr7ZY525ilcsdo-XHlhh/view?usp=sharing). In this video, the robot has its joint reference angle feed in at 80hz, the plot shows the roll angle of the robot, the blue curve indicates the desired roll angle and the red circle indicates the IMU reading data. I think the walking doesn't shake that much but the reading shakes. How am I supposed to use this noisy data to regulate the lateral balance of the robot, do you have any suggestion? (without any regulation on lateral balance, the robot has a great chance to fall sideway so that it can not walking for a long distance.)
2. I think the OP3 has the default balance algorithm, right? I have read the code and I think the default balance algorithm use the IMU data. But did you guys just use this noisy raw data to regulate the balance? 
3. Does op3 also have pitch angle regulator implemented as the default balance algorithm?

ROBOTIS-Kayman, 2019-12-18:
Hi,

- We use only gyro value after low pass filter to balance OP3. How about using gyro instead of IMU?
- OP3 has 2 walking algorithm.
   - [op3_walking_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_walking_module)  
      It has simple balance control using gyro. (applied to roll axis and pitch axis.)  
      Please refer to the [source code](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/b2d35a1f545e3db6bb196f4b649dcbc6303e0efb/op3_walking_module/src/op3_walking_module.cpp#L886).
   - [op3_online_walking_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_online_walking_module), [op3_balance_control](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_balance_control)  
      This algorithm is more complicated than the on above. It is divided into a walking and a balance control. Looking at the [upper process function](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_balance_control/src/op3_balance_control.cpp#L675} may help you.

I hope this helps your research.

Regards,
Kayman

yuan0623, 2019-12-18:
Kayman
Thanks so much for your instruction. I would like to try the simple version first.
Here is my understanding: 
- the desired body angular velocities at roll and pitch axes are all 0, thus we have following definitions `double rl_gyro_err = 0.0 - sensors["gyro_x"]; 
    double fb_gyro_err = 0.0 - sensors["gyro_y"];`
- then the  `sensoryFeedback(rl_gyro_err, fb_gyro_err, balance_angle)` is used to compute the angles which are required fot balance the robot.

However, I do come some following questions.
- in your previous message, you have mentioned using a low-pass filter to filter the angular velocities,. However, I don't see this part in the code. Does the gyro have the embedded low-pass filter to filter data automatically? Or I have to implement the filter myself after I've subscribed the imu data.
- in the following source code:

```
{
 for (int idx = 0; idx < 14; idx++)
    {
      double goal_position = 0.0;
      if (get_angle == false && idx < 12)
        goal_position = goal_position_.coeff(0, idx);
      else
        goal_position = init_position_.coeff(0, idx) + angle[idx] + balance_angle[idx];

      target_position_.coeffRef(0, idx) = goal_position;
}
```
I assume that `idx` is the motor number. As the balance angles become effective only if  `14>idx>12`, should I understand it as the balance angles only work for two joints: `r_knee` and `l_knee`. But if they only work on knee joints, how does the robot balance itself the in roll axis?

ROBOTIS-Kayman, 2019-12-20:
You understood well.

- The low-pass filter is processed in the sensor module.   
    The value you use inside the process function is a filtered value.
    - [open_cr_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/b2d35a1f545e3db6bb196f4b649dcbc6303e0efb/open_cr_module/src/open_cr_module.cpp#L114)
- idx is the joint index, not motor id.
    [joint index in the source code](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/b2d35a1f545e3db6bb196f4b649dcbc6303e0efb/op3_walking_module/src/op3_walking_module.cpp#L57)

Regards,
Kayman

yuan0623, 2019-12-20:
Thanks for your response. I really appreciate your patience and help. However, I still have the following questions:
- if idx is the joint index, then the balance angle only be applied when 12<idx<14, which means the only joint 12 `(r_sho_pitch)` and joint 13 `(l_sho_pitch)` is used to balance, right?
- Is there any specific reason for you guys to use gyro to balance the robot rather than the orientation of the robot, in my opinion, the orientation provides more detail information, right?

ROBOTIS-Kayman, 2019-12-20:
- `if (get_angle == false && idx < 12)` means don't apply the balance when the leg's goal angle can not be got.
- I didn't develop the original of this algorithm, so I don't know in detail, but it seems to have taken this way because of the lack of reliability of the sensor.

Regards,
Kayman

closed 2020-01-27 by yuan0623