ROBOTIS-GIT/ROBOTIS-OP3 #87: op3/both walking and grasping

xxchenchen, 2020-01-05:
The direct control module is activated to capture the object from the ground and the object is captured. Then we move to the walking position. Walking with modulating. We run the object to the target we want to drop/leave . To put the object on the target, we need to switch back to the Direct control module. But in the case, the robot puts all its motors in the starting position. That's when he drops the object on the floor. He sits down quickly. We can't drop the object on target.

ROBOTIS-Kayman, 2020-01-06:
I need your answers to the questions below to help you.
- Does it continue to appear every changing the module? 
- What if you changed the module in different situations?
- Does it happen in all joints?
- Are you updating your goal position in other modules?
    When changing the module, motion_module reads the goal position saved in dxl status and moves based on it. 

Regards,
Kayman

xxchenchen, 2020-01-07:
Hi Kayman,
Thanks for answer. T
1 -Does it continue to appear every changing the module?

When you move from the Walking Module to the DirectControl Module, the robot releases the object. There is no problem switching from DirectControl to Walking.

2- What if you changed the module in different situations?

There is problem when we pass to walking module from direct control module or action module.

3-Does it happen in all joints?
Changing to the direct control module from walking module,  all joints take the starting position.

4-Are you updating your goal position in other modules?
We have different goals for each module. For example, in the Walking module the goal is to walk for 3 seconds. In the DirectControl Module, the target will drop the object to the table we are in front of after 3 seconds.

ROBOTIS-Kayman, 2020-01-08:
Hi,

The goal position I mentioned means the goal position of dynamixel. It is a value to put as result variable in process function of each module. 
The direct_control_module and action_module move based on this value, so that each joint can move to 0 degrees when changing the mode If the goal position is initialized.

Please refer to these codes.
 - [direct_control_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_direct_control_module/src/direct_control_module.cpp#L253) : updating result from goal position of dxl
 - [robotis_controlelr](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/master/robotis_controller/src/robotis_controller/robotis_controller.cpp#L1299) : making syncwrite data to dxl using result variables

Regards,
Kayman

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will