ROBOTIS-GIT/ROBOTIS-OP3 #88: Walking program

fuad99, 2020-01-11:
can i make own walking program using kinematics algorithm?

ROBOTIS-Kayman, 2020-01-28:
Hi,

Sorry for the late response.
Yes, you can make the walking program.
If you use our [framework](http://emanual.robotis.com/docs/en/software/robotis_framework_packages/), you can make other walking module like [op3_walking_module](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#op3_walking_module) and [op3_online_walking_module](http://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#op3_online_walking_module). and we also provide ROS Package(op3_kinematics_dynamics) for kinematics.
- source code 
  - [op3_walking_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_walking_module)
  - [op3_online_walking_module](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_online_walking_module)
  - [op3_kinematics_dynamics](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/tree/master/op3_kinematics_dynamics)

If you don't want to use our framework, you should make walking program using [DYNAMIXEL SDK](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/overview/)

Regards,
Kayman

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will