ROBOTIS-GIT/ROBOTIS-OP3 #94: NUC

robotiscustomer, 2020-04-06:
Hi Kayman, I want to buy a NUC to learn how to use OpenCR, ROS and various Robotis platforms. I cannot find information about the model of the NUC used in the OP3. The specs only say Intel NUC i3, dual core, 8GB DDR4 RAM 2133MHz and 128 SSD.  Can you tell the model number of that NUC? If buying now, which NUC do you recommend? Don't know if any NUC and iGPU is compatible with your products.

ROBOTIS-Kayman, 2020-04-06:
Hi,

The model name of NUC used in OP3 is nuc 7i3BNK. It was changed to the above model in the second half of 2018, before that it was the NUC6i3SYK.
It may change in the future depending on the revision of NUC.

Other models are not compatible with the frame of the OP3, and even in the same model, power line work is required to put it in the OP3.

Regards,
Kayman

robotiscustomer, 2020-04-06:
Thanks Kayman. What suggestion do you have in this case? Sounds like I need to get the NUC7i3BNK to avoid compatibility issue? I don't know if your ROS software libraries use some specific instruction sets such as AVX and may be Intel GPU is required. Can you please provide more information about the required power line work?

ROBOTIS-Kayman, 2020-04-06:
Hi, 

Any NUC can be used on our platform, but certain platforms may require more computing power.
Two of our products are using NUC. One is THORMANG3 and the other is OP3. In the case of THORMANG, i5 is used because it requires more computing power. 
OP3 libraries and framework may work regardless of CPU or GPU, but we can not gaurantee because we didn't test other SBCs.

About power line, refer to #52 please.

Regards,
Kayman

robotiscustomer, 2020-04-07:
Hi Kayman,

Good morning. Thank you for your reply.

In #52, it was mentioned that your company would soon make a manual about the wiring and made it available.  I cannot find it. Where is it?

ROBOTIS-Kayman, 2020-04-07:
Hi again,

We tried to release the manual with the kit type OP3, but, unfortunately, the plan has changed and we do not provide a manual.

Regards,
Kayman

robotiscustomer, 2020-04-09:
Hi Kayman,

Thank you. 

In #59, you posted a diagram to show that one leg is connected directly to a RS485 port of the OpenCR. Then, another leg and two arms are each connected to a separate RS485 port of a hub. What are the reasons for doing that?

onjinwook, 2020-04-09:
Hi,

because ... It can solve the problem of "Voltage Drop". That's why we do like that.

Regards,
Jin

robotiscustomer, 2020-04-09:
Hi Jin,

Thank you. So, if I make a biped or hexapod later, it is better to split the limbs and connect each limb separately to an individual RS485 connector of a hub the way shown in #59 ?

onjinwook, 2020-04-09:
Hi again.

Yes. that is better. 
and if you have a plan for the hexapod, you have to do.

Regards,
Jin.

robotiscustomer, 2020-04-09:
Thanks Jin.

In the case of the OP3, when only the LB-012 battery is used, how does the battery power the entire robot? Is it like LB-012 (11.1V, 1800mAh)-->Power barrel of the NUC--usb-->OpenCR->20 dynamixels?

I am amused that the LB-012 can power both the NUC, OpenCR and 20 dynamixels.

From the photo in #59, power source for the entire robot seems to be connected to the OpenCR barrel (the red line). There is no mention about how the SBC/NUC gets the power. I am a bit confused. Can you clarify please?

onjinwook, 2020-04-09:
Hi,

You can check the issue from "[#52](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/issues/52)"

Regards.
Jin.

robotiscustomer, 2020-04-11:
Thanks Jin.

It seems to be:

LIPO--(T-connector on OpenCR)-->OpenCR & Dynamixels-- (SMW250-02 on OpenCR)-->Intel NUC--(USB port of NUC)-->U2D2--(RS485)-->OpenCR.

Am I correct? I am a bit confused about the connection between the U2D2 and other components. Given that we can connect the Dyamixels from the OpenCR, why we need a U2D2?

onjinwook, 2020-04-13:
Hi, 

Of course .. you make a another system if you want. 
That is another way.
But OpenCR has some functions - external power, sensing, and so on.
we just want to separate those - Dynamixel control & ETC
That is why we use a U2D2.

Regards.
Jin

robotiscustomer, 2020-04-26:
Thanks. I have been thinking about this but I am still a bit confused. 

As far as I understand, OpenCR gets power from the LIPO battery. It then supplies power to the NUC and the 20 dynamixels. From the communication map diagram in #59, it seems that the right leg is connected directly to the OpenCR but the left leg, two arms and the head are connected to the PR28-EXT hub. Where is the hub connecting to? Another RS485 port on the OpenCR or the U2D2? 

Can you clarify please?

onjinwook, 2020-04-28:
Hi. 

As you can see, You may find the 3 RS485 ports on the OpenCR.
[image]
but we need 5 points.
so we use a PR28-EXT hub. it can connect 4 points.
To conclude, PR28-EXT hub connect to RS485 port.

Regards.
Jin

robotiscustomer, 2020-04-28:
Thanks Jin. So although the U2D2 is in the above diagram and those in #52, it actually does not do anything?

onjinwook, 2020-04-29:
Hi, 
SBC is using this device(U2D2).
if you are programming in ROS(in NUC), 
you can send commands through this device

Regards.
Jin

robotiscustomer, 2020-05-05:
Thanks Jin. I am very sorry but I am still a bit confused. So, if I use SBC such as Raspberry Pi, I can send commands through the U2D2 to the Dynamixel. In the case of the OP3 which uses ROS and NUC, the U2D2 is not used? Also, for the hub that merges the limbs and the neck to the OpenCR, is it a passive hub (i.e. the GND, VDD, D+ and D- of the limbs and the neck just physically connect to each other and no power source is connected to the hub)?

onjinwook, 2020-05-07:
Hi,
OP3 uses U2D2 for controlling Dynamixel.
and power is connected to the OpenCR(Battery or Powersupply)
There's(-OpenCR-) both power and communication is connected.

Regards.
Jin.

robotiscustomer, 2020-05-10:
Thanks Jin. Do you mean in addition to connecting different branches of limbs to the hub, I also need to supply additional power to the hub?

onjinwook, 2020-05-14:
Hi,
and sorry for the late answer.
If you have already power on OpenCR, you don't need power anymore.
check below.

[image]

robotiscustomer, 2020-05-21:
Thanks Jin.

It looks like in the OP3, the NUC's case is removed but for the Thormang, the two NUC were installed on the robot as is from retail boxes. When using only the NUC's motherboard to control a robot, what to do with the metal frame which came with the NUC? It seems to be used for some kind of shielding. Does removing it cause some kind of issues?

onjinwook, 2020-05-27:
Hi,
sorry for late reply. 
We don't have any problems with removing the case.

Regards.
Jin.

robotiscustomer, 2020-06-01:
Thanks Jin. I successfully removed the case.

closed 2020-12-11 by ROBOTIS-Will