ROBOTIS-GIT/ROBOTIS-OP3 #95: OP3 Online Walking on Gazebo: RLEG IK ERR

tahara-study, 2020-05-07:
Hi, OP3 developers.

I'm using OP3 and an external Desktop PC for my development.
In order to simulate Online Walking on Gazebo, I executed the following commands on the external PC.

roslaunch op3_gazebo robotis_world.launch 
roslaunch op3_gui_demo op3_demo_walking.launch 
roslaunch op3_manager op3_manager_gazebo.launch

When I press "Balance On" bottom or "Balance Off" bottom from the OP3 Demo GUI app,
the following error is displayed in op3_manager op3_manager_gazebo.launch.

[ INFO] [1588829400.131925649, 23.453000000]: [START] Balance Gain
[ WARN] [1588829400.134447106, 23.455000000]: RLEG IK ERR : -3
[ INFO] [1588829400.134588242, 23.455000000]: [FAIL] Task Space Control

In contrast, op3_manager op3_manager_gazebo.launch worked correctly when it was executed on the real OP3 robot.

My question:
Can't the inverse kinematics calculations required for Online Walking be performed on an external PC?

Thank you very much.

Full results of op3_manager op3_manager_gazebo.launch
--------------------------------------------------------------
NODES
  /
    op3_manager (op3_manager/op3_manager)

ROS_MASTER_URI=http://Ubuntu18.local:11311

process[op3_manager-1]: started with pid [15271]
[ INFO] [1588829376.514794976]: manager->init
[ WARN] [1588829376.524727483]: SET TO GAZEBO MODE!
/dev/ttyUSB0 added. (baudrate: 2000000)
[PortHandlerLinux::SetupPort] Error opening serial port!
(/dev/ttyUSB0) [ID:  1]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  2]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  3]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  4]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  5]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  6]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  7]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  8]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:  9]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 10]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 11]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 12]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 13]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 14]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 15]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 16]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 17]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 18]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 19]     XM430-W350 added. 
(/dev/ttyUSB0) [ID: 20]     XM430-W350 added. 
(/dev/ttyUSB0) [ID:200]        OPEN-CR added. 
[ WARN] [1588829376.576837863]: Fail to load offset yaml.
[ WARN] [1588829376.884217642]: Head control - angle unit : 35
[ERROR] [1588829376.909037783]: Fail to load offset yaml file.
[ INFO] [1588829376.909087074]: parse pose for moving init pose
[ WARN] [1588829376.910345669]: Get the init gain from Dxl init file
[ INFO] [1588829377.016594493]: Go to init pose
[ INFO] [1588829377.018019391]: Base Module is enabled
[ INFO] [1588829380.361423303]: Base Module is enabled
[ INFO] [1588829394.179689057, 18.164000000]: [START] Joint Control
[ INFO] [1588829394.191027120, 18.171000000]: [start] send trajectory
[ INFO] [1588829397.589367112, 21.157000000]: [END] Joint Control
[ INFO] [1588829398.714248351, 22.174000000]: [end] send trajectory
[ INFO] [1588829400.131925649, 23.453000000]: [START] Balance Gain
[ WARN] [1588829400.134447106, 23.455000000]: RLEG IK ERR : -3
[ INFO] [1588829400.134588242, 23.455000000]: [FAIL] Task Space Control

closed 2020-06-22 by tahara-study

AlfretzNehemia, 2022-04-12:
I have same problem, RLEG IK ERR : -3, the value from rleg_err always < 0. How to fix that? Thank you

tahara-codes, 2022-04-12:
Hi.
As a simple solution, it works on Ubuntu16, ROS kinetic.
It doesn't work on Ubuntu18 or Ubuntu 20 in my experience.