ROBOTIS-GIT/ROBOTIS-OP3 #98: Issue running op3_bringup

lachg258, 2020-08-21:
Hello, I am currently trying to run the tutorials from the e-manual for the ROBOTIS OP3. 

I am doing section 5.1 of the tutorials. I run the sudo command to kill the demo service. Although when I run the command:
$ roslaunch op3_bringup op3_bringup.launch  
I get the following error.
[image]
[image]

Can you please help me with this issue, I am not sure if I am not running an appropriate file or what the issue is. Thank you!

ROBOTIS-Will, 2020-10-06:
Hi @lachg258,
I'm sorry about the late response.
The first warning indicates that the camera package cannot find the calibration file which is not necessary for the OP3 default demo. If you need to utilize camera for specific visual operation, please calibrate the camera.
The deprecation warning doesn't affect to the operation, but warns you that there will be deprecated feature in the camera package.
Thank you.

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will