ROBOTIS-GIT/ROBOTIS-OP3 #101: Additional Configuration for Custom OP3

luqmansen, 2020-10-05:
Hello @ROBOTIS-Will,

Currently, I'm my team build a clone of OP3 with a slightly different configuration and facing quite a lot of problems, I there any additional configuration that needed to be changed in order to achieve the same functionality as OP3? We have changed every single servo configuration from .robot file and servo offset

When we trying to run the op3 demo, no error showed on the console, but no single servo is moved. 

Specs
- NUC Intel Celeron (lower specs, but shouldn't be a problem)
- OpenCR (Same)
- Camera (Logitech C920)
- Upper Body (6 * MX-28)
- Lower Body (12 * MX-64)

We've been stuck for quite a while. Any help advice would be very helpful, thank you in advance

[image]

ROBOTIS-Will, 2020-10-06:
Hi @luqmansen ,
Thank you for sharing the picture of your robot :+1: 
This OP3 package is designed specifically for the OP3 hardwares and therefore could require significant modifications depending on your hardware configuration.

First of all, please check if proper OP3 Arduino sketch is downloaded on the OpenCR.
Please refer to [Arduino IDE section](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#arduino-ide) of OpenCR1.0 emanual for installing required board manager.
Once the OpenCR board manager is installed correctly, open Arduino IDE and find the OP3 example in the OpenCR example list.

Secondly, if your MX DYNAMIXEL is using Protocol 1.0, it is recommended to update the firmware to Protocol 2.0.
In case using Protocol 1.0 which is a default of MX series, you'll need to modify the [Protocol column in OP3.robot](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/OP3.robot) file.

Also, make sure to correctly set the initial value settings in the [dxl_init_OP3.yaml](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_manager/config/dxl_init_OP3.yaml) file.

I'd recommend to operate each joint with DYNAMIXEL Wizard 2.0 first then try to control with your software.
Thank you.

luqmansen, 2020-10-06:
Thanks for your response @ROBOTIS-Will  however, we have done what you've listed on your answer but still no luck. However, we still have quite a doubt about this [MX-64.device](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/master/robotis_device/devices/dynamixel/MX-64.device) file (and Mx-28 one).

The provided configuration is using the addressing table for protocol 1.0, however, we notice this and change all the addressing table to protocol 2, since this file is required on OP3.robot. 

Can you help with this? thank you in advance.

ROBOTIS-Will, 2020-10-07:
@luqmansen 
Oh, I see. The current MX-64 and MX-28 device files are written based on Protocol 1.0.
You may still use these files if you are going to use default firmware of MX series.
When using Protocol 2.0, these device files also needs to be updated and you can refer to [XM430](https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/blob/master/robotis_device/devices/dynamixel/XM430-W210.device) device file which is very similar to Protocol 2.0 MX DYNAMIXEL. 
Please refer to [product emanual](https://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/) for updating the control table values and you may not needed to change the `torque_to_current_value_ratio` and `velocity_to_value_ratio` as these are optional when running DYNAMIXEL with Current or Velocity control modes.

Please note that the file name should be the same as the `MODEL` name of the OP3.robot file (e.g. XM430-W350 refers to XM430-W350.device file)

Thank you.

ROBOTIS-Will, 2020-12-11:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2020-12-11 by ROBOTIS-Will