ROBOTIS-GIT/ROBOTIS-OP3 #103: Converter TTL to RS485 for dynamixel

fuad99, 2020-10-21:
is it possible to combine R series dynamixel with T series dynamixel? because i'm using dynamixel 2XL430-W250 to add the DoF in robotis OP3. can the TTL dynamixel convert to RS485? i'm very need help :) 
Thanks!

ROBOTIS-Will, 2020-11-03:
Hi @fuad99 ,
Under the limited condition, TTL and RS-485 DYNAMIXEL can be mixed, but when Sync / Bulk instructions are used like OP3's case, 2XL430-W250 is hard to be connected to XM430-W350-R.
If you use TTL port on U2D2 and OpenCR, then modify the code to separately control the 2XL430-W250, it could be a viable option.
[image]

fuad99, 2020-11-03:
I've done adding .device file on ROBOTIS-FRAMEWORK after i add 2XL430-W250 on OP3.Robot. But while i run op3_manager, error info appear "Robotis first bulk fail". it's problem with bulk sync? 

Today, i try the test program inside the DYNAMIXEL-SDK, and it's work to control Dynamixel 2XL430-W250 and XM430-W350 together! 
The problem i've found is declaration of port(just declaration, both is '/dev/ttyUSB0'). If i put same declaration port at 2 Dynamixel, it will be error. But if i declared diffrent port like this:
port1='/dev/ttyUSB0' (for XM430-W350) 
port2='/dev/ttyUSB0' (for 2XL430-W250) 
it's work to bulk read/write.

The next problem is i can't find this solution in Robotis-Framework. 
Is it still possible to edit Robotis Framework?

ROBOTIS-Will, 2020-12-11:
Hi,
Since TTL and RS-485 are not recommended to put together and therefore we do not have further instruction on these configuration.
Robotis-Framework is an open source and you can freely modify upon your necessity.
Thank you.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.

closed 2025-03-18 by ROBOTIS-zerom