ROBOTIS-GIT/ROBOTIS-OP3 #104: Why can't I find this file?

Echo334, 2020-11-09:
[image]

ROBOTIS-Will, 2020-11-09:
@Echo334 
I'm sorry about the delayed reply.
Could you specify the OP model you have? Is it OP3?
Also, did you change/upgrade any component?
Thank you.

Echo334, 2020-11-09:
@ROBOTIS-Will 
Thank you for your reply. In fact, our robot model is OP2.  We haven't changed or updated its components. There's no doubt in my mind that the problem is the USB camera is not properly connected to the PC. So what should we do to solve this problem?

ROBOTIS-Will, 2020-11-27:
@Echo334 
The error clearly shows that the USB camera is not detected by the system.
Could you try below command and paste the result here?
```bash
$ ls /dev/video*
```

Echo334, 2021-01-04:
@ROBOTIS-Will 
I'm sorry about the delayed reply.In fact,when I connect again, I couldn't connect to Putty and Vnc suggested an  error which is 'Failed to connect to sever'.

ROBOTIS-Will, 2021-01-04:
Are you trying to upload the OP3 software on OP2?
OP2 has its own ROS software and OP3 packages may not compatible with OP2.
Please use OP2 packages as instructed in the eManual below.
https://emanual.robotis.com/docs/en/platform/op2/ros_package/#robotis-op2-ros-packages

Echo334, 2021-01-10:
Sorry, I can't do as you said, because I can't connect to the OP2 terminal window at the moment.From the current situation, I think the problem is that the USB camera is not properly detected by the system.Is there any possibility of camera damage?

ROBOTIS-Will, 2021-01-11:
You can directly connect the monitor and input devices to OP2 instead of connecting via ssh on terminal.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.