ROBOTIS-GIT/ROBOTIS-OP3 #106: Walking Speed increased in OP3 Soccer Demo

lachg258, 2020-12-11:
Hi guys, 

I have noticed that when the ROBOTIS OP3 is in the "Autonomous Soccer Mode", when walking he seems to walk at a faster rate when a particular range from the ball, then proceeds to slow down when it is in close proximity to the ball. I have had a search but can not seem to find where in the code this process is created? Could you please highlight what section this is in as I would like to edit the speed the OP3 walks when it is heading towards the tracked soccer ball. 

Thank you!

ROBOTIS-Will, 2021-02-04:
Hi,
The walking motion is controlled by the op3_walking_module package.
In order to modify the walking behavior, you can tune parameters.
Please refer to below emanual for detailed instructions.
https://emanual.robotis.com/docs/en/platform/op3/robotis_ros_packages/#op3_walking_module

The source code of following the ball can be found from `op3_demo/src/soccer/ball_follower.cpp` in the ROBOTIS-OP3-DEMO package.
Thank you.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.

closed 2025-03-18 by ROBOTIS-zerom