ROBOTIS-GIT/ROBOTIS-OP3 #107: Gazebo SImulation Problem

full0134, 2020-12-18:
Hello sirs,

I am trying to get the gazebo simulation working but it seems to run into an issue. The simulation loads but then does not work. I have attached the terminal outputs below. Sorry if this is an obvious problem, as I am quite new to using ROS/Gazebo.

Gazebo:
robotis@robotis-op3 ~ $ roslaunch op3_gazebo robotis_world.launch 
... logging to /home/robotis/.ros/log/50b22eaa-40ec-11eb-b989-645d86d0e2d3/roslaunch-robotis-op3-4927.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://10.42.0.1:42283/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /robotis_op3/head_pan_position/joint: head_pan
 * /robotis_op3/head_pan_position/pid/d: 0.0
 * /robotis_op3/head_pan_position/pid/i: 0.0
 * /robotis_op3/head_pan_position/pid/p: 5.0
 * /robotis_op3/head_pan_position/type: effort_controller...
 * /robotis_op3/head_tilt_position/joint: head_tilt
 * /robotis_op3/head_tilt_position/pid/d: 0.0
 * /robotis_op3/head_tilt_position/pid/i: 0.0
 * /robotis_op3/head_tilt_position/pid/p: 5.0
 * /robotis_op3/head_tilt_position/type: effort_controller...
 * /robotis_op3/joint_state_controller/publish_rate: 50
 * /robotis_op3/joint_state_controller/type: joint_state_contr...
 * /robotis_op3/l_ank_pitch_position/joint: l_ank_pitch
 * /robotis_op3/l_ank_pitch_position/pid/d: 0.0
 * /robotis_op3/l_ank_pitch_position/pid/i: 0.0
 * /robotis_op3/l_ank_pitch_position/pid/p: 50.0
 * /robotis_op3/l_ank_pitch_position/type: effort_controller...
 * /robotis_op3/l_ank_roll_position/joint: l_ank_roll
 * /robotis_op3/l_ank_roll_position/pid/d: 0.0
 * /robotis_op3/l_ank_roll_position/pid/i: 0.0
 * /robotis_op3/l_ank_roll_position/pid/p: 50.0
 * /robotis_op3/l_ank_roll_position/type: effort_controller...
 * /robotis_op3/l_el_position/joint: l_el
 * /robotis_op3/l_el_position/pid/d: 0.0
 * /robotis_op3/l_el_position/pid/i: 0.0
 * /robotis_op3/l_el_position/pid/p: 10.0
 * /robotis_op3/l_el_position/type: effort_controller...
 * /robotis_op3/l_hip_pitch_position/joint: l_hip_pitch
 * /robotis_op3/l_hip_pitch_position/pid/d: 0.0
 * /robotis_op3/l_hip_pitch_position/pid/i: 0.0
 * /robotis_op3/l_hip_pitch_position/pid/p: 50.0
 * /robotis_op3/l_hip_pitch_position/type: effort_controller...
 * /robotis_op3/l_hip_roll_position/joint: l_hip_roll
 * /robotis_op3/l_hip_roll_position/pid/d: 0.0
 * /robotis_op3/l_hip_roll_position/pid/i: 0.0
 * /robotis_op3/l_hip_roll_position/pid/p: 50.0
 * /robotis_op3/l_hip_roll_position/type: effort_controller...
 * /robotis_op3/l_hip_yaw_position/joint: l_hip_yaw
 * /robotis_op3/l_hip_yaw_position/pid/d: 0.0
 * /robotis_op3/l_hip_yaw_position/pid/i: 0.0
 * /robotis_op3/l_hip_yaw_position/pid/p: 50.0
 * /robotis_op3/l_hip_yaw_position/type: effort_controller...
 * /robotis_op3/l_knee_position/joint: l_knee
 * /robotis_op3/l_knee_position/pid/d: 0.0
 * /robotis_op3/l_knee_position/pid/i: 0.0
 * /robotis_op3/l_knee_position/pid/p: 50.0
 * /robotis_op3/l_knee_position/type: effort_controller...
 * /robotis_op3/l_sho_pitch_position/joint: l_sho_pitch
[... 152 lines cut ...]
(/dev/ttyUSB0) [ID:  5]         XM-430 added. 
(/dev/ttyUSB0) [ID:  6]         XM-430 added. 
(/dev/ttyUSB0) [ID:  7]         XM-430 added. 
(/dev/ttyUSB0) [ID:  8]         XM-430 added. 
(/dev/ttyUSB0) [ID:  9]         XM-430 added. 
(/dev/ttyUSB0) [ID: 10]         XM-430 added. 
(/dev/ttyUSB0) [ID: 11]         XM-430 added. 
(/dev/ttyUSB0) [ID: 12]         XM-430 added. 
(/dev/ttyUSB0) [ID: 13]         XM-430 added. 
(/dev/ttyUSB0) [ID: 14]         XM-430 added. 
(/dev/ttyUSB0) [ID: 15]         XM-430 added. 
(/dev/ttyUSB0) [ID: 16]         XM-430 added. 
(/dev/ttyUSB0) [ID: 17]         XM-430 added. 
(/dev/ttyUSB0) [ID: 18]         XM-430 added. 
(/dev/ttyUSB0) [ID: 19]         XM-430 added. 
(/dev/ttyUSB0) [ID: 20]         XM-430 added. 
(/dev/ttyUSB0) [ID:200]        OPEN-CR added. 
[ INFO] [1608267598.045100598, 20.028000000]: Load offsets...
[ WARN] [1608267598.397615243, 20.038000000]: Head control - angle unit : 30
[ INFO] [1608267598.699967199, 20.047000000]: parse pose for moving init pose
[ WARN] [1608267598.704118912, 20.047000000]: Get the init gain from Dxl init file
[ INFO] [1608267598.846664266, 20.050000000]: Go to init pose
[ INFO] [1608267598.865311984, 20.050000000]: Base Module is enabled
[ INFO] [1608267598.950302469, 20.051000000]: [start] send trajectory
[ INFO] [1608267602.956757226, 20.151000000]: [end] send trajectory

ROBOTIS-Will, 2021-02-04:
Hi @full0134 
I'm sorry about the delayed response, but I don't see a particular issue from your log.
If you see the [image] icon instead of Pause [image] icon at the bottom of Gazebo screen, Gazebo is not running, but waiting to be played.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.