ROBOTIS-GIT/ROBOTIS-OP3 #109: how to change the frame of the robot's leg?

tzf230201, 2021-02-16:
I changed the frame of the robot's thighs and legs to be slightly longer mechanically, this doesn't really matter when using the action module, but it affects the waking module, I want to ask, which part of the program should I change, in order to fit the new length of the frame robot leg?

ROBOTIS-Will, 2021-02-17:
Hi @tzf230201 
I'm sorry, but we do not provide a thorough consultation regarding product customization.
If you have modified the frame, you will need to modify walking parameters, balance control, kinematics dynamics, and motion data.
Thank you.

tzf230201, 2021-02-17:
No, that's enough to answer my question, thank you very much

tzf230201, 2021-02-19:
I am trying to understand op3_kinematics_dynamics,

I'm having a little trouble understanding where the values ​​in op3_link_data come from.

it looks like the value is different from the urdf file.

can any one tell me where these values ​​come from or are there any references so i can understand them.

    /* ----- body -----*/

    // pelvis_link
    op3_link_data_[29]->name_ = "pelvis";
    op3_link_data_[29]->parent_ = 28;
    op3_link_data_[29]->sibling_ = -1;
    op3_link_data_[29]->child_ = 19;
    op3_link_data_[29]->mass_ = 6.869;
    op3_link_data_[29]->relative_position_ = robotis_framework::getTransitionXYZ(0.0, 0.0, 0.0);
    op3_link_data_[29]->joint_axis_ = robotis_framework::getTransitionXYZ(0.0, 0.0, 0.0);
    op3_link_data_[29]->center_of_mass_ = robotis_framework::getTransitionXYZ(-0.011, 0.000, 0.058);
    op3_link_data_[29]->joint_limit_max_ = 100.0;
    op3_link_data_[29]->joint_limit_min_ = -100.0;
    op3_link_data_[29]->inertia_ = robotis_framework::getInertiaXYZ(0.03603, 0.00000, 0.00016, 0.02210, 0.00000,
                                                                    0.03830);

    /* ----- head -----*/

    // head_pan
    op3_link_data_[19]->name_ = "head_pan";
    op3_link_data_[19]->parent_ = 29;
    op3_link_data_[19]->sibling_ = 1;
    op3_link_data_[19]->child_ = 20;
    op3_link_data_[19]->mass_ = 0.087;
    op3_link_data_[19]->relative_position_ = robotis_framework::getTransitionXYZ(0.0, 0.0, 0.0205);
    op3_link_data_[19]->joint_axis_ = robotis_framework::getTransitionXYZ(0.0, 0.0, 1.0);
    op3_link_data_[19]->center_of_mass_ = robotis_framework::getTransitionXYZ(0.000, -0.002, 0.010);
    op3_link_data_[19]->joint_limit_max_ = 0.5 * M_PI;
    op3_link_data_[19]->joint_limit_min_ = -0.5 * M_PI;
    op3_link_data_[19]->inertia_ = robotis_framework::getInertiaXYZ(0.00011, 0.00000, 0.00000, 0.00003, 0.00000,
                                                                    0.00012);

    // head_tilt
    op3_link_data_[20]->name_ = "head_tilt";
    op3_link_data_[20]->parent_ = 19;
    op3_link_data_[20]->sibling_ = -1;
    op3_link_data_[20]->child_ = -1;
    op3_link_data_[20]->mass_ = 0.724;
    op3_link_data_[20]->relative_position_ = robotis_framework::getTransitionXYZ(0.0, 0.0, 0.03);
    op3_link_data_[20]->joint_axis_ = robotis_framework::getTransitionXYZ(0.0, -1.0, 0.0);
    op3_link_data_[20]->center_of_mass_ = robotis_framework::getTransitionXYZ(0.009, 0.046, 0.022);
    op3_link_data_[20]->joint_limit_max_ = 0.5 * M_PI;
    op3_link_data_[20]->joint_limit_min_ = -0.5 * M_PI;
    op3_link_data_[20]->inertia_ = robotis_framework::getInertiaXYZ(0.00113, 0.00001, -0.00005, 0.00114, 0.00002,
                                                                    0.00084);

Thank you in advance

bryntirion2, 2021-03-03:
look for the file kinematics.h

bryntirion2, 2021-03-03:
in the op3_walking_module.cpp

  double thigh_length = 93.0 * 0.001;  //m
  double calf_length = 93.0 * 0.001;  //m
  double ankle_length = 33.5 * 0.001;  //m
  double leg_length = 219.5 * 0.001;  //m (THIGH_LENGTH + CALF_LENGTH + ANKLE_LENGTH)

ROBOTIS-Will, 2021-03-04:
Hi, as described in the source code, those values are for OP2, and the computeIK is not used for OP3 anymore.
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/blob/master/op3_walking_module/src/op3_walking_module.cpp#L236

tzf230201, 2021-03-05:
so where do the op3_link_data [] values ​​come from?

tzf230201, 2021-03-05:
because I can't understand where these values ​​come from.
for a while I'm using kinematics with classical algebraic methods instead.

and it works

but it would be nice that I can use the method that robotis programmer has worked on very beautifully

closed 2021-03-07 by tzf230201