ROBOTIS-GIT/ROBOTIS-OP3 #115: The Robotis OP3 has a stuttering on his servo

dnovx, 2021-07-17:
I'm having a problem with my OP3 robot, the problem is when the robot does a motion or walk, the servos on my OP3 robot always have seizures or stutter, and when I touch the robot, the robot moves for a few seconds like in the video I attached in this issue thread

https://user-images.githubusercontent.com/11470383/126029918-f68d4608-9784-42df-9f61-c64796e3300f.mp4

ROBOTIS-Will, 2021-07-19:
Hi @dnovx 
When operating the OP3, please connect the battery.
Connecting the SMPS power supply will prevent discharging the battery, but the SMPS contact could be loosen during motion and may need to be backed up with the battery.
Thanks.

dnovx, 2021-07-19:
I've used the genuine battery from Robotis OP3 instead of using the SMPS Adapter. but the result is the same, the servo experiences it like stuttering. and will move in a few seconds when given a vibration / touch on the robot.

Is there any other solution?

ROBOTIS-Will, 2021-07-20:
@dnovx 
Was the battery fully charged? The genuine battery is charged less than half when shipped due to the safety regulation.
The symptom could be caused by the communication issue between the controller and DYNAMIXEL.
Please check if all cables are undamaged and firmly connected to each DYNAMIXEL.

You may also try the OP3 recovery with provided NUC 7 image
http://localhost:4000/docs/en/platform/op3/recovery/#recovery-of-robotis-op3

dnovx, 2021-07-20:
@ROBOTIS-Will Thanks for your reply sir.
I've done OP3 Recovery by reinstalling everything like Linux Mint and its ROS using a USB Flash Drive from ROBOTIS but the result is still the same. and I've also swapped U2D2 followed by swapping Mini PC and OpenCR from another robot but the result is still the same.

should we replace all the RS485 cables contained in the servo?

ROBOTIS-Will, 2021-07-21:
@dnovx 
If the same issue can be reproduced with replaced PC, OpenCR and U2D2, the problem could be the RS485 cables or DYNAMIXEL as it seems Goal Position is not successfully transmitted to DYNAMIXEL.
I'd recommend replacing the RS485 cables and see if it resolves the problem.
Thanks

dnovx, 2021-07-21:
@ROBOTIS-Will 
How to find out which part of the RS485 cable on the servo is the problem and then replace the new RS485 cable? because when I checked using Dynamixel Wizard 2.0 sometimes all servos were detected but a few seconds later nothing was detected and even OpenCR was not detected.

ROBOTIS-Will, 2021-07-21:
@dnovx 
When checking with DYNAMIXEL Wizard 2.0, make sure to terminate other programs such as demo program.
https://emanual.robotis.com/docs/en/platform/op3/getting_started/#how-to-kill-the-demo-program

The OpenCR OP3 firmware will try to communicate with connected DYNAMIXEL which may interrupt communication with the Wizard 2.0 so it is recommended to disconnect the communication cables(3: `D+`, 4: `D-`) while connecting the power.
[image]

dnovx, 2021-07-22:
@ROBOTIS-Will 
for that I have done it by first disabling the running Demo Mode. and then I tried to scan using Dynamixel Wizard 2.0 and it was detected. But only a few seconds something happened like there was a disconnected RX TX connection from the servo

after that I re-scanned and the results were all servo and OpenCR were not detected.

But I still don't know where the problem is from which servo or from which RS485 cable.

ROBOTIS-Will, 2021-07-23:
You might want to check out this thread for searching the bad harnessing.
https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/issues/114#issuecomment-884191491

dnovx, 2021-07-23:
@ROBOTIS-Will 
Thank you very much for the last solution sir. finally the robot has recovered as usual. Thankfully, the robot is no longer stuttering.

https://user-images.githubusercontent.com/11470383/126850957-66d9c7d6-f3bb-4552-9e7f-099adcfcf397.mp4

ROBOTIS-Will, 2021-07-26:
@dnovx 
Awesome! Glad to see that the OP3 is running well.
Was the bad cable the cause of problem?

dnovx, 2021-07-26:
@ROBOTIS-Will 
Yes, I found some of the cables inside the servo, like they were chipped and almost broke. 
Thanks sir

ROBOTIS-Will, 2021-07-27:
Thank you for the information.
Your information is greatly helpful to resolve similar issues.
Stay safe and have a great day!

closed 2021-10-09 by dnovx