ROBOTIS-GIT/ROBOTIS-OP3 #117: ROS installation error while installing the humanoid_navigation package

janson97, 2021-09-22:
Hello there, 
I have been struggling to get past this installation step and cannot identify what the issue is, it is only when I use the catkin_make command with the humanoid_navigation module. I get the following error. 

[image]

I've tried downgrading the libpcl-dev package but still nothing. Have been stuck for weeks, it's a thesis project and I am unable to move forward. I've attached a full output below as well as results for "locate uniform_sampling"

[Full Output Error.txt](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/7210776/Full.Output.Error.txt)
[Locate_results.txt](https://github.com/ROBOTIS-GIT/ROBOTIS-OP3/files/7210786/Locate_results.txt)

ROBOTIS-Will, 2021-09-27:
hi @janson97 
The PCL version 1.7.2 has a different file structure and does not have the uniform_sampling.h under the `pcl/filters`, but it is in the [pcl/keypoints](https://github.com/PointCloudLibrary/pcl/tree/0ddf11a6ad1175b6f1c253da003de20c1295cde0/keypoints/include/pcl/keypoints)

If you need to use the older version of PCL, you may need to modify the HumanoidLocalization.cpp so that the correct directory is referred.

Thank you.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.

closed 2025-03-18 by ROBOTIS-zerom