ROBOTIS-GIT/ROBOTIS-OP3 #120: How to let the op3 raise raise hand when walking.

leongsheng, 2022-05-21:
How can I change the parameter of sho_pitch in walking module. Since I would like to let the op3 raise hand when walking. Or it's have another easier way to do this?

ROBOTIS-Will, 2022-05-31:
Hi @leongsheng 

The robot walking requires well balanced motion. Raising arm while walking could shift the CoG of the robot that may critical in balanced walking.
The easiest approach to create a specific motion is to use the [action editor](https://emanual.robotis.com/docs/en/platform/op3/tutorials/#how-to-create-the-motions).
This allows you to create motions for demo.

Otherwise, you may need to heavily modify and tune the walking algorithm which we are not able to provide technical support.

Thank you.

ROBOTIS-zerom, 2025-03-18:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.

closed 2025-03-18 by ROBOTIS-zerom