ROBOTIS-GIT/ROBOTIS-OP3 #121: question about torque limits of OP3 motor

yuan0623, 2022-08-05:
Hi, I just want to ask the torque limit about the motor on Darwin. I have checked the manual, the manual tells me that the stall torque is 2.5Nm and 4.1 Nm for MX-28 and XM430. I'm not sure how stall torque related to the torque limit. Also, can you explain to me how gear ratio affect to the torque limit?

Best
Yuan

shiba-8ro, 2022-08-10:
@yuan0623 
Hi,The Dynamixel's stall torque is the maximum available, but not a practical number. If stall torque is generated and left for a while, the motor may burn. So, apart from that, it is necessary to explicitly set the torque limit with a practical numerical value.

ROBOTIS-zerom, 2025-03-18:
Hello! Thank you for your question about the torque limits of the motors on the ROBOTIS-OPs. I’ll explain the relationship between stall torque and torque limit, as well as how the gear ratio affects the torque limit, based on the information from the manual.

- Stall Torque
Stall torque is the maximum torque a motor can produce when it is fully stopped (stalled) under maximum load. According to the manual, the stall torque is 2.5Nm for the MX-28 and 4.1Nm for the XM430. This value reflects the motor’s electrical output and mechanical design (particularly the gear ratio). However, operating at stall torque for extended periods can lead to overheating or damage, so caution is advised in practice.

- Torque Limit
The torque limit is a software-defined maximum torque that the motor is allowed to output, adjustable via the Dynamixel control table’s Torque Limit (address 34, 2 bytes). It ranges from 0 to 1000 (0 = 0%, 1000 = 100%), capping the motor’s maximum torque output, including the stall torque. For example, setting the torque limit to 500 allows the motor to use only 50% of its maximum capacity.

- Relationship
Stall torque represents the motor’s physical maximum capability, while the torque limit controls the output within or below that maximum. The torque limit can be set to a value less than or equal to the stall torque, allowing users to adjust it for safety or energy efficiency. For instance, if the MX-28’s stall torque is 2.5Nm, a torque limit of 1000 (100%) permits up to 2.5Nm, but a limit of 500 (50%) restricts it to 1.25Nm.

The gear ratio is the reduction ratio of the gear train, adjusting the torque and speed transmitted from the motor’s output shaft. For example, a gear ratio of 200:1 reduces the motor’s input speed by 200 times while increasing torque by a factor of 200 (ignoring efficiency losses). A higher gear ratio amplifies the motor’s initial torque (the torque produced by the motor itself), resulting in greater torque at the output shaft.

Stall torque is the motor’s physical maximum output, while the torque limit is a software control within that range. The gear ratio amplifies the motor’s initial torque to determine the stall torque, with higher ratios providing more torque at the cost of speed.

ROBOTIS-zerom, 2025-03-25:
This issue will be closed due to the lack of customer response. If you have any further questions or encounter this issue again, please feel free to reopen this issue or create a new one.

closed 2025-03-25 by ROBOTIS-zerom