ROBOTIS-GIT/ROBOTIS-OP3 #125: How can I use multiple robotis op3 under the same network with one master?

cobiansm, 2023-06-08:
I’ve been trying to use two Robotis OP3 under the same network connected to a single master (my computer), but even with namespaces and tf_prefix, they are copying each others actions and I cannot control them individually, how can they be separated?

cobiansm, 2025-03-19:
Thank you for responding, we solved by adding a robot id ros param to each topic.

On another note, we’d like to ask if there’s a way to get the robot’s odometer with a built-in method?

ROBOTIS-zerom, 2025-03-25:
The default ROBOTIS-OP3 software does not include a direct feature of obtain odom data in the form of `nav_msgs/Odometry` messages. This is because the OP3's locomotion method does not involve wheels, and the primary purpose of the basic software focuses on providing fundamental robot motion control and perception functionalities.

ROBOTIS-zerom, 2025-03-31:
This issue will be closed as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2025-03-31 by ROBOTIS-zerom